richo/homeassistant-franklinwh #1: HACS integration? (feature request)

bradyr, 2024-03-18:
Greetings -- I am curious (...and very interested as a fellow Franklin-WH system owner) if there are plans to integrate this with HACS in Home Assistant so that I can start pulling data from my aGate in Home Assistant.

richo, 2024-03-18:
Hi! I'm kinda new to the HA ecosystem, but I can hopefully fill in some gaps. I wound up having to continue development in private for a bit while I figure some things out with Franklin, but I can tell you rn that I have it working well enough to both pull data from the aGate about load, power consumption, and totals, etc and also control smart switches and so on.

I'm new to HA though and so I dont' know what HACS is? Right now it just presents as a bunch of sensors.

bradyr, 2024-03-18:
HACS is the "community store" for 3rd party integrations and add-ons for Home assistant.  it's like a friendly way for users to connect home assistant to 3rd party repo's for integrations like what you're working on here.

That said -- right now i'm just looking for exactly what you're talking about:  a bunch of sensors that pull all the data from the aGate (grid/solar/battery/generator).  being able to toggle the smart circuits is nice but i'm more interested in the sensor data so i can "finally" get a accurate power setup in my home assistant.

i'd be happy to help test some of this out in private with you if need be.

richo, 2024-03-18:
Ah I see!! Yes, once it's stable I will be happy to figure out how to get it up in the store. If you just want the sensor data, I will try to backport all of that stuff into the public repository. Are you ok with installing it as a custom_integration, or would you prefer I get it up into the store?

bradyr, 2024-03-18:
I'd be OK with doing a custom integration for myself.  i could use the learning experience anyways.  If time permits, have a look at the HACS store to see if their method of doing custom integrations is of any value to you.

that all being said, I'm just excited that someone was actually able to get this info from the aGate into home assistant.

richo, 2024-03-19:
Ok! I'll try to get this updated tonight. It should suffice to just clone this repo into the custom_integrations folder, that's how I have it running in my hass instance.

bradyr, 2024-03-20:
Sounds great -- looking forward to testing it out.  thanks again for taking the time to work on this.

richo, 2024-03-23:
Hey, haven't forgotten about this, but backporting everything has turned into more work than I expected. I'll @ you when I get it figured out.

richo, 2024-03-23:
Actually, I think offhand, what's in this repo as it stands should give you a working set of sensors to describe some of the home state, you will need to get an access token though, I can actually get the script that fetches them up tonight, it's not as entanged as most of the other stuff.

[image]

richo, 2024-03-23:
Ok so. It's a little bit convoluted but bare with me.

I just pushed a login script to https://github.com/richo/franklinwh-python/blob/main/bin/login.py

Run that with your email and password as arguments, and it'll give you a login token. From there you should just need to clone this repo into `custom_integrations/` in your homeassistant install, then restart hass (I couldn't figure out a better way to force it to install dependencies and load everything).

You need to add a config stanza to your configuration.yaml like:

```yaml
sensor:
  - platform: franklin_wh
    access_token: TOKEN
    id: ID
```

Where TOKEN is the token the login script gave you, and ID you can get from the app by going to the hamburger menu, More, Site Address, it's nex to the sn: in Device address.

Sorry this is a bit jank. I've made quite a bit of progress in a private branch, but I can't release that for a little bit. Thanks agian.

@bradyr

bradyr, 2024-03-23:
@richo good morning -- I was able to get this working and i'm pulling data from my aGate into my home assistant instance.   It looks like the default update interval is 15 mins in the latest config in this repo.   unless you already have some updated config with 1 min (live), 1 day (and maybe 1 month) intervals, I'm going to attempt to build out those sensors so that i can fully integrate my agate and battery into the home assistant energy "setup".   

thanks again though, this is awesome.

richo, 2024-03-23:
Morning Ryan,

Glad to hear it's working! When I'm able to release the new version, the
data will be pretty close to live. It also is pushed from the agate rather
than polled from hass.

I chose the 15 mins thing fairly arbitrarily, it's a one line change to
make it 1 min if you want to do that.

Richo

bradyr, 2024-03-23:
I was able to tweak the update interval from 15*60 to just 60, as a test and it worked.

I wonder if we can pull historical or daily data off the aGate (kWh information like the aPower charge and discharge energy information) or if i'll need to create kWh conversions in home assistant and monitor it that way.  the energy dashboard in Home assistant likes usage over time information, not real time data.  just random thoughts...

richo, 2024-03-23:
There is an API for that! I uh, potentially do not have the stomach to
implement it right now, but I can share the details with you if you'd be
interested.

My original plan for this was just to give hass a window in time and let it
do logging and stuff for historical data, but if you want to grab that it
is available. The trick will be that I think hass sensors regard their past
states as being absolute, so you'll wind up with historical data that maybe
isn't the most useful.

richo

bradyr, 2024-03-23:
i think i'll stick with home assistant pulling "live" data from the aGate and building its own history, which i can (should be able to?) build energy-over-time sensors from (kWh).   that sounds a lot easier than pulling specific point-in-time historical data from the aGate

closed 2024-03-28 by bradyr