richo/homeassistant-franklinwh #2: access token question

npdsomerhayes, 2024-05-23:
Hi 

I can't work out how to generate the access token is it possible to give a more in-depth explanation?

richo, 2024-05-23:
Yeah totally, sorry this is all pretty rough. I published a script here https://github.com/richo/franklinwh-python/blob/main/bin/login.py

Running it with your username and password should spit out an access token.

richo, 2024-05-23:
Let me know how you go. I'll be pushing a fairly significant update sometime around June 7th, hopefully that helps with a lot of the setup pain points.

npdsomerhayes, 2024-05-23:
I have the access token I am heading away for a week so I will finish the Home assistant integration once I am home.

I do have a question regarding the configuration.yaml example you have provided. 
Regarding the sensors do I use the example or do I need to provide a name, ID, units etc? or is it pulling this information from the sensor.py code?  

I will look out for the update and thanks for all your work on this.

richo, 2024-05-26:
I think from memory it gets it from the code.

I think in the other thread I included a configuration snippet and it's
pretty thin. If you have other things that would be useful to you please
open an issue and I'd be happy to add them.

richo, 2024-05-28:
@npdsomerhayes I'm at a computer now so I can answer your questions better :)

This is an example of one of the sensors: https://github.com/richo/homeassistant-franklinwh/blob/main/sensor.py#L155-L172

So the unit, name, etc is all defined there and you shouldn't need to set it in config. I believe it's possible to override them in your configuration.yaml but to be honest I am not a HA expert, I just kinda needed this to work and so I've been figuring things out as I go along.

npdsomerhayes, 2024-05-30:
Thanks Richo I had a play around today but I am getting the same error as slackwilson is having. My system is based in New Zealand and has the 1006 serial number also. I will wait for your next update to see if that fixes it. 

I also agree with slackwilson request to be able to change between modes "Emergency backup", Self Consumption" and "Time of Use"

closed 2024-05-31 by npdsomerhayes