richo/homeassistant-franklinwh #3: Trouble installing

slackwilson, 2024-05-25:
Hi, franklinwh user here. 
Would love to get this working on my home assistant, have tried to work it out myself before logging an issue but struggling.

I managed to run the login script to get a token and recorded that.. FYI I did this on my main windows PC, is that ok or will I need to run that on home assistant system for it to work? 

Anyway I obtained a token above and system ID from the mobile app under site address (I also noted that my id starts with 1006 not 1005) I added this to my existing sensor section of configuration.yaml

sensor:
  - platform: filter
    name: "filtered watertank ADC voltage"
    entity_id: sensor.watertank_adc
    unique_id: sensor.watertank_filtered_adc
    filters:
      - filter: lowpass
        time_constant: 15
        precision: 2

  **- platform: franklin_wh
    access_token: :eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.exxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
    id: 1006xxxxxxxxxxxxxxx**
 
I also copied the files from this repository into a folder created called franklin_wh under custom components

[image]

In the logs I get the following error for each of the sensors.. 

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:950
First occurred: 1:48:25 PM (29 occurrences)
Last logged: 2:02:25 PM

Update for sensor.franklinwh_state_of_charge fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 950, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1295, in async_device_update
    await hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/franklin_wh/sensor.py", line 96, in update
    self._attr_native_value = stats.current.battery_soc
                              ^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'current'

I am sure I have done something stupid or missed something, can you confirm the folder structure is OK, and see what I may have done wrong?

Also, would love to help with this project, even if it's something boring like documenting how to install for people like myself lol. 

What I want ultimately is to be able to see information that I think you already have setup, and in terms of ability to control the main thing I would love is to be able to change the Mode - ie change between the 3 modes in the app, "Emergency backup", Self Consumption" and "Time of Use"

Any help is greatly appreciated.

richo, 2024-05-25:
Hey Thanks for reaching out.

Running the script on your windows PC shouldn't make a difference, the tokens are all the same.

What you have looks about right to me, it might be a bug on my end, or they might have changed something since it's not exactly a stable API. Let me have a look and get back to you.

I would absolutely love some help documenting. Anything you can contribute would be very appreciated.

As far as changing the modes, I haven't done the work to reverse engineer that yet, but if you would use it I can add it sometime soon, no worries.

slackwilson, 2024-05-25:
Awesome, thankyou. 

no problem at all helping with documentation, can't guarantee the quality but I'll do my best :) good excuse to play with github too so might watch some vids/read up on how to contribute then make a start. 

Changing modes would be something I would absolutely use.  I only had the battery installed recently, its approaching winter here in Australia and I'm already finding just using an app limiting and far too manual, for example if its a cloudy day and by a certain time of day if the battery is still low charge, I can save money by changing mode to time of use so it draws from the grid until 3pm which is start of peak cost. However on days with good sun, it charges enough I don't need to draw anything from the grid during the day, so self consumption works best to avoid unnecessarily drawing from grid. Seems like something I should be able to script in Home Assistant but would need that ability to change modes to make it useful, otherwise best I can do with only sensors would be to send me an alert on my phone to tell me to change the mode in the app.

richo, 2024-05-25:
That makes sense, It's a similar usecase to what I am using it for (Charging my gf's electric car selectively when our batteries are full).

I'll do my best to get back to you in the next day or so when I'm able to chase up why you're getting that error.

bradyr, 2024-05-25:
@slackwilson Good morning -- I've also integrated this project into my home assistant instance and so far it's been working flawlessly.   The only change i made to the code was the update interval:

file: sensor.py, line 58, changed to:
`UPDATE_INTERVAL = 60`

here's my configuration.yaml snippet for the franklin sensor(s):
`# initial franklinWH testing:
sensor:
  - platform: franklin_wh
    access_token: ######################
    id: 1005###############
`

yes, my aGate also has a 1005 id -- I do not know if 1006 is a different hardware revision of the unit but we can dive deep into that by comparing our setups and if there is any different information provided in the official franklinWH android/iphone apps.  my aGate is fully outfitted with the "smart" circuits (which aren't so smart imo), generator module and aPbox module.  my aGate is also hardwired to my network -- not cellular or wifi (if that matters)

-ryan

slackwilson, 2024-05-26:
@bradyr  Any help, information & comparison is gladly received!

Here is my home assistant version, almost identical just a slightly different version for the supervisor. 
```
Core 2024.5.5
Supervisor 2024.05.1
Operating System 12.3
Frontend 20240501.1
```
I don't know if the ID: is different due to being in Australia and 240v or if it's simply a newer model as it was only installed a coupe of weeks ago so it's very new and could well be a slightly newer model.

I don't have any smart circuits configured, but I do have a generator module (although interestingly it doesn't show in the app). I assume it's the same for you guys where the installer seems to have a different level of control (he showed me his app and it looks different) 
Here is a screenshot of my App version
 
[image]

and I have triple checked the sn: just to make sure I'm not crazy
[image]

bradyr, 2024-05-26:
@slackwilson @richo 

yeah pretty sure franklin changed something on their end with regards to generating an access token from the https://energy.franklinwh.com/hes-gateway/terminal/initialize/appUserOrInstallerLogin URL.

as a test, i re-ran the script to generate my token and it produced a radically different result than the original token.  I might suspect that the token is random but this output is radically different than before.

sure enough when i tried replacing my "working" token with the "new" token, i get no data from the aGate.  if i roll back to my previous access token, it works fine.

i have a feeling there is some kind of conversion that needs to be done on the new output from the python script, but i'm no python expert.

anyways ill mess around with it on my end and see if i can figure out if i can hack it back to working.

richo, 2024-05-26:
Interesting. I do know that they are planning an update soon so this might
be a part of that, I will make some time this week to see what's going on
with it.

I'm hoping it's just a change to the format, shouldn't be too hard.

Thanks for the updates this is helpful.

slackwilson, 2024-05-26:
@bradyr 
Still with my thinking it's something obvious I am missing , What home assistant install are you running? 

I have the simple HA OS, installed on an old intel NUC device. 

What I thought of, is how is the python running? I'm wondering if my basic HA OS install does not understsand the .py files at all. 

Do you have any python add-ons?

bradyr, 2024-05-26:
@slackwilson 
I'm running HA OS on a older dell optiplex desktop -- no special config other than using the beta builds when they get released.   I did not have to install any dependencies for this integration.   

I suppose its possible that some dependencies were already met due to other 3rd party integrations that I've installed, but i have a feeling that HA OS has the python support already included since it's a critical component of the system.

I'm pretty sure the root cause is going to be related to the TOKEN value being generated by the helper script.  As i mentioned in an earlier post, I generated a fresh token as a test (...the new token looked similar to yours) and it broke my integration.  I didnt go looking for the exact error that you got, but when i rolled back to my previous token the sensors immediately worked again.

the token output from the helper script needs to be modified in some way in order to work.   if you try running the script with a bad password, it wont generate any output so it's "almost" working.  maybe it's choking on the length of the "new" token now -- i don't know.

slackwilson, 2024-05-26:
@bradyr 
Ok thanks heaps for confirming that, at least I know our HA installs are near enough to identical. 

I wasn't sure about python as when I google running python in HA I get people talking about installing add-ons like pyscript or app-daemon so wondered if maybe you already had something like that installed.

richo, 2024-05-28:
@slackwilson ok I had a bit of a look at this. I think franklinwh have started on the updates I eluded to elsewhere (I need to coordinate a rollout around june 7 for what I believe will be some breaking changes).

With that said, do you have a token with a leading `:`? If so, I believe just removing it will probably make things start working.

richo, 2024-05-28:
Sorry I also missed a couple of other comments, HA has a preference for HA modules that import external modules and have very little functionality in them themselves, so the bulk of the franklinwh logic is over here: https://github.com/richo/franklinwh-python/blob/main/franklinwh/client.py

However, for sort of annoying reasons I wound up pulling some stuff over into HA as a stopgap. On or around June 7 I should be able to release some updates that improve a whole bunch of things, one of them being that the code will be a lot less of a mess. The short version is I have a bunch of extra functionality in some private branches and merging it into the public stuff is a pain for right now.

slackwilson, 2024-05-28:
@richo  Yes me token starts with a :  and funny enough I did wonder if that was simply a form of electronic punctuation so tried it with and tried it without, both have same result, I have taken it back out now that I know it shouldn't be there. 

So I have not installed/copied or done anything with your https://github.com/richo/franklinwh-python/tree/main/franklinwh repository - do I need to copy / install that into HA for things to start working?

Or am I just better off waiting for you to get a chance to to your updates around June7 ?  I'm keen as mustard to get this working, and also to help with doco and whatever I can but if it's easier for you to focus on a bit of cleanup/update whatever 1st I'll wait :)

richo, 2024-05-28:
Yeah that's new. You shouldn't need to do anything with my other repo, the manifest in the custom_integration should download and install it all by itself ( and @bradyr managed to get it working so clearly it **can** work).

Interesting that it still doesn't work even removing the extra `:`. I just fetched a token from there and was able to use it with one of my test scripts, so it looks like their API is definitely accepting it.

Ok, I bet I know what the issue is. The old token was simple enough that I think yaml would let you use it as a bare string, the new format probably needs to be wrapped in `""`.

so you should have something that looks like

```yaml
sensor:
  - platform: franklin_wh
    access_token: "______________________:____________________________________________________________________________________________________________________________________________"
    id: 1005____________
```

If that doesn't work, can you get me the error you see in the logs and I'll do my best to debug remotely. It's kinda up to you how much you want to wait, I am sort of lowkey hoping that not much will need to change in your configuration.yaml but some of the entities might get replaced when I do the update. It should give you things like basically realtime updates though. I also haven't forgotten about the mode switching thing, although honestly I likely won't be able to look at it for a little bit.

I wouldn't stress too much about your ID starting 1006, while I am curious I think everything more or less has to support the same protocol. Thanks for being patient!

slackwilson, 2024-05-28:
Ok after removing the leading : I have also wrapped the token in ""

here is snippet of configuration now with plenty of characters _'d out

```
  - platform: franklin_wh
    access_token: "eyJ0e______________________________9.e______________________________________________________________________0.f________________________________________5o"
    id: 10060006A__________

```

Here is the list of errors  I see in system logs, 

[image]

this is drilled into one - the one for state of charge . They all look to be the same, but happy to copy each of them

```
Logger: homeassistant.helpers.entity
Source: helpers/entity.py:950
First occurred: 3:41:53 PM (2 occurrences)
Last logged: 3:42:23 PM

Update for sensor.franklinwh_state_of_charge fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 950, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1295, in async_device_update
    await hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/franklin_wh/sensor.py", line 96, in update
    self._attr_native_value = stats.current.battery_soc
                              ^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'current'
```
Let me know if you need something from another area. 

Also where as in what folder location could I check to make sure the custom integration has downloaded itself ok to make sure that's worked and not broken? 

Obviously I don't want to post it in public, but I'm also ok with privately working with you to check my system, even possibly sending you my ID & Token so you can test it on your system etc

richo, 2024-05-28:
Ok! Thanks for all the help. I think I see what I've done. The fix might be a little annoying, I am testing it locally and then I'll push it up and post some instructions.

bradyr, 2024-05-30:
@npdsomerhayes @richo 

i noticed this morning that i was no longer getting data from my aGate so i took a look and i see that Rich has been tweaking some code, trying to figure out the issue.  I think this is great.

to the point though, i was able to get data coming back from the agate by changing my token to the following in my configuration.yaml file:

`"APP_ACCOUNT::_______random_stuff___________________.long_string_of_random_chars.____random_stuff____________________"`

basically i generated a new token, which spit out that really long string.   I copied the entire string with the "APP_ACCOUNT::" at the start of it (if you're not seeing the APP_ACCOUNT::" in the output, then add it in so it matches the format above.   wrap it in double quotes, restart HASS and pray.

so far so good for me right now.

TheSwert, 2024-05-30:
Wanted to also chime in and confirm prefixing the "APP_ACCOUNT::" to the token resolved my connectivity issue as well.

@richo , thank you for the hard work on this integration.

richo, 2024-05-30:
Morning folks. I'm glad that's resolved it for now. What I have in place is
pretty jank but I should be able to roll something more permanent out in th
next couple of weeks. Glad it's working though, I was hoping to update this
thread today.

npdsomerhayes, 2024-05-30:
@bradyr that solved it thanks everyone for your help

slackwilson, 2024-05-30:
@npdsomerhayes yours is working now? 
I have obtained the new token which is super long and starts with the APP_ACCOUNT:: copied all that and wrapped it in "" but I still have the same errors :(

npdsomerhayes, 2024-05-31:
Yes mine is working

I am not sure if this matters but I did generate my token from my home assistant (HA) server as this token is different to the token generated by my Mac. I have not tried the token  I generated on my Mac but can if you want me to?

to run the login.py script from my HA server I had to manually install 'requests' as this was not pre-install on my version of HA 

`pip3 install requests`

below is my configuration.yaml

`sensor:
  - platform: franklin_wh
    access_token:  "APP_ACCOUNT::_______random_stuff___________________.long_string_of_random_chars.____random_stuff____________________"
    id: 1006****************
`
let me know if you need any more info on my setup I am no expert in HA or Python I just know enough to get by! but happy to help if I can.

Probably a little off-topic!
My network plan gives me 3 hours of free power from 9:00 PM to midnight at the moment I need to switch between self-consumption to Time of use and then back again in the morning.
I read your comments at the beginning regarding mode changes from Time of use and self-consumption. I think there is a bug in the time-of-use setup as in the advanced settings you can set how you want the system to behave but it does not work. I spoke to the supplier of the system and they talked to FranklinWH apparently there is a plan to separate New Zealand and Australia from the US configuration and they are hoping to fix this bug.

richo, 2024-05-31:
The tokens will be random now I believe. @slackwilson are you on the latest
version?

richo, 2024-05-31:
On your scheduling thing, yes I should be able to let you do that fairly
easily. Right now I have my system setup to turn on my gfs car charger when
our solar production starts feeding the grid, the logic for your thing
should be similar but driven by the time instead of charge state.

slackwilson, 2024-05-31:
@npdsomerhayes if you don't mind getting a token from your Mac and checking that also works?  

I am running HA OS, it doesnt seem to have a full operating system and I'm not really sure how to run a python script from a command line on it, I have installed terminal & ssh addon, but the terminal doesnt recognise any of the usual pip3 or python commands.

If your mac token works for you, then that will confirm I don't really need to get the token from my HA system anyway so I can eliminate that. 

@richo I will download the login.py from the bin folder of (https://github.com/richo/franklinwh-python)  again to be sure and re-run, this (https://github.com/richo/homeassistant-franklinwh) repository looks unchanged though, but I can also redownload and copy this one too if you think it might help?

richo, 2024-05-31:
Oh I might have pushed my changes only to the python module. I think I need
to bump the manifest. In theory I think a clean install would pull the
latest from pypi anyway tho.

npdsomerhayes, 2024-05-31:
@slackwilson yes my Mac generated token works also

slackwilson, 2024-05-31:
@npdsomerhayes Awesome thankyou for testing that, I am getting a token from my desktop so at least I check that off the list as a possible problem, to be fair @richo did say it should be fine earlier in this thread, and obviously it is. 

@richo if I delete the franklin_wh folder under custom components, reboot then recreate the folder and copy the files back in again, is that sufficient to "reinstall" ?

npdsomerhayes, 2024-05-31:
@slackwilson it might be worth posting your configuration maybe one of us can spot what is wrong?

richo, 2024-05-31:
Honestly I'm not sure. I would say probably. I can push the manifest change
when I get home in an hour or two, but if you want to try in the meantime
please let me know how it goes!

slackwilson, 2024-05-31:
Happily, and help is greatly appreciated:

My configuration.yaml

```
sensor:
  - platform: franklin_wh
    access_token: "APP_ACCOUNT::exxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxn0.lgDKOu9xGiW4IxxxxxxxxxxxxxxxxxxxxdP8LXnu-bo"
    id: 10060006Axxxxxxxxxx
```

And my folder structure - please check this, I have the files from this repository directly in the folder I created named "franklin_wh" under custom components, no sub folders, no other files from any other repository are in there (IE I don't see anything extra downloaded automatically in there)

[image]

richo, 2024-05-31:
That looks correct to me.

npdsomerhayes, 2024-05-31:
The only difference I can see is that  I have an extra folder "__pycache__" that contain two additional files.
ignore the login.py I put that in there when I generated the token from my HA server.

[image]

slackwilson, 2024-05-31:
Ah yes I have that now also, I took that screenshot after recreating and downloading the files but now it also has the __pycache__ folder

[image]

@richo how can I confirm it is auto-downloading the correct files in the background? is there a folder somewhere I can go and check it has downloaded the files from https://github.com/richo/franklinwh-python

npdsomerhayes, 2024-05-31:
@slackwilson One other thing that might be worth checking is the permissions of the files. I have encountered this with another add-on where I copied the files using a file explorer instead of downloading them directly to the HA folder form the HA terminal.

[image]

bradyr, 2024-05-31:
@slackwilson i agree with what @npdsomerhayes is saying regarding file permissions.  i also cloned the repo thru the HASS terminal directly and all the files are owned by root:root u should be able to chown them thru the terminal or temporarily rename the folder and re clone the repo as root.

slackwilson, 2024-05-31:
ok, I deleted again, did a git clone this time from the terminal add-on, renamed the folder to franklin_wh and confirmed root is the owner - pretty sure it was before, but went through the steps just to be 100%

see screenshot of my steps and confirmation of owner

[image]

npdsomerhayes, 2024-05-31:
@slackwilson if you give me a secure way (email/whatsapp...) to send you my token and serial number I will be happy to send them to you to test in your configuration at least then you will know if it is a token/serial number problem or not.

slackwilson, 2024-05-31:
@npdsomerhayes thats very generous thankyou 

lets try this temporary email mandarin51@eamsurn.com I just created - send an email to that, and I'll reply with my normal email address which we can than send private info.. I'm happy to also give you my token & serial number so you can try it

richo, 2024-05-31:
> lets try this temporary email ***@***.*** I just created -

I'm just walking in the door now and I can push that manifest change. I can
also put up a standalone script to test credentials.

slackwilson, 2024-05-31:
I have same issue trying it with @npdsomerhayes token and id, so it looks like its not a token/id issue but something with my HA setup?
I have also given my token and id to @npdsomerhayes so he can try if he wants to

richo, 2024-05-31:
Huh, that is so bizarre. If you want to send me the token at richo@psych0tik.net I can try to see if I can spot what is wrong.

I guess one thing I forgot to mention earlier is that at present, franklinwh doesn't have any mechanism to revoke these authentication tokens, which is why I am kinda resistant to shipping them around, there's no way for you to revoke that access when you have given it, essentially.

slackwilson, 2024-05-31:
@richo @npdsomerhayes 
Thankyou both, so @npdsomerhayes was able to use my token and id successfully, turns out even though I had checked multiple times I kept missing a number on my ID which he picked up. 

So current status is @npdsomerhayes HA can connect to my id/token OK, and of course his. 

I cannot get my HA to work with my id/token, or with @npdsomerhayes id/token.. 

I think it's safe to say my issues are isolated to my HA system, and it's not a token issue.

richo, 2024-05-31:
Ok that helps to narrow it down. I will push a new version with some more
logging to hopefully track down where the issue is. Thanks for the update.

slackwilson, 2024-05-31:
ok, another update.. in desperation, I shutdown my main HA system, and ran up a brand new fresh HA install in vmware workstation. The only things I installed were terminal&ssh, file editor, and I git cloned [the](https://github.com/richo/homeassistant-franklinwh) using the exact same steps in my screenshot above. 

Result: it connects!

which confirms 100% my ID & token are fine, and also I guess it confirms my steps to install are ok too. 

So that leaves perhaps remnants from previous attempts, I certainly have deleted the franklin_wh folder under custom_components a couple of times but will do it again

@richo is there any other folder location I can clean up? I asked before but not sure if you missed it or it was a stupid question lol, is there is another folder or files located somewhere that get auto-downloaded I can check & delete?

Failing that, I can only assume some other add-on or integration is interfering with this, or something about my configuration is an issue.

bradyr, 2024-05-31:
@slackwilson good morning -- you might have already gone down this troubleshooting path, but since the franklin integration is working on your "test" HA install/VM -- i would start adding in your other custom integrations one by one on the test instance to see if/when it breaks.

i cant imagine it being a dependency issue but who knows at this point.

richo, 2024-05-31:
My intuition is that it was the FranklinWH module (separate from the home
assistant addon) not updating itself and your fresh install forced it to
grab the latest. I don't know enough about HA to tell you where to look to
verify that though, unfortunately.

richo, 2024-05-31:
I did push the update last night though so now the latest source on main
depends explicitly on a version that should work.

slackwilson, 2024-05-31:
@bradyr   Yep I considered the options, and thought maybe I'd work on the main system 1st and I'd start with the easiest elimination steps, so on my main HA system I stopped/disabled a bunch of add-ons in batches of 2 or 3, and at one point.. it started working on my main HA system!  then I started re-enabling add-ons and it stopped working again. 
At thought point I thought ah-ha! it must be one of these add-ons is conflicting, So turned them back off again one at a time.. ended up with them all off again, and it still wasn't working.. which I couldn't wrap my head around, 

I walked away in disgust and left it.  I came back maybe 2 hrs later, and it was working!!
I could see the errors in the logs had about 17 instances or something which at least confirmed I wasn't crazy it was failing /not working when I walked away, then at some point it just decided to work on its own. 

After that I re-enabled the add-ons one at a time, got them all back on except for zigbeeMQTT which has an error, but I installed ages ago that in preparation  for getting zigbee devices but don't actually have any yet.  

At this point, its still working this morning and I can only assume @richo is right, it was the module and it maybe updated itself?  It was several hours after richo did updates I think so I guess it might add up timing wise if it was a few hours delayed in updating. Or maybe one of the add-ons was stopping it from updating.. no idea. 

The only other thing it could have been is a conflict with zigbeeMQTT, and interestingly the very first time it started working I had disabled a few add-ons together and in that batch it included Mosquito broker (MQTT) and zigbeeMQTT, probably a co-incidence but I'm really not sure. 

I'm keeping an eye on it, but apart from a couple of update errors it's been working fine overnight.

Big thankyou to all, especially @npdsomerhayes helping eliminate token/id as the issue, and picking up my mistake with the id, and thanks to everyone for their ideas and input.

I'll close off this issue I think, and open a fresh one for the changing of modes so it's in a clean thread

closed 2024-05-31 by slackwilson

richo, 2024-06-01:
For what it's worth I have Mqtt stuff running and mine works ok. I'm glad
your setup is working now! I will try to make the errors more helpful when
things aren't right moving forward.