richo/homeassistant-franklinwh #4: Change battery mode?

slackwilson, 2024-05-31:
Would be really awesome if we could add the ability to change mode, specifically change between the 3 operating modes of the system - "Emergency backup", Self Consumption" and "Time of Use".

This is something I would absolutely use - an ability to use home assistant automation to change between the three modes for smarter energy savings than what the franklin app currently provides. 

It would be a massive improvement over manually changing on the app for times when you can save money by switching between the modes during the day.

richo, 2024-06-03:
Thanks for filing the issue so this doesn't get lost. Just to set expectations, i won't be able to touch this until that release comes out, and I am off to a race this week, so I would expect to see some movement sometime next week. If your installs break because of upstream changes I apologise in advance, I will be doing my best to update things when I get back.

slackwilson, 2024-06-03:
Thanks Richo

Out of curiosity, I was looking at the responses.rs file in this repository of yours https://github.com/richo/franklin-wh/tree/main/src

I can see a listing of what look like data from the API,  some of them, purely based on guesswork from the short names might be handy, for example in the home assistant energy dashboard which I use, for battery info it wants kWh inputs, which it looks like that might be available, and of course I spotted the one called "mode" which is probably what I'm after. 

Anyway I was wondering if there was a quick & dirty way to dump all that raw info? that way I could confirm what does what.. IE if I could dump the info and check what Mode is showing, then go an change the mode in the app and dump the info again.. that way I could at least test and confirm what fields are handy and which ones aren't

I am trying to wrap my head around your scripts, and it looks like the client.py file in the franklinwh module only grabs a select set of data currently?

Anyway just curious if there was a uick & dirty method to check api data, otherwise completely understandable you have other things in life to worry about, and I'll patiently wait :)

richo, 2024-06-03:
> I can see a listing of what look like data fields from the API, some of

Yeah absolutely. Yeah, I wrote the rust implementation first and then after
finding out that hass really only wants python implementations I did a
python one.

You're correct that there's a ton of stuff in there, unfortunately that API
is read only so yes you can absolutely do the thing you're talking about,
but that won't get you closer to changing the mode from hass. I can put up
a script that simply dumps everything in that json blob sometime tonight
though.

richo, 2024-06-19:
Please excuse all the issue spam, but i'm just trying to keep this straight for myself. As part of my work on #8 I grabbed enough stuff to implement this as well. I wanted to check a couple of things though:

1. Are you content to just be able to change the modes, or do you need to change the TOU plan etc as well? Right now it looks like changing the mode and the hedge % is easy, doing the TOU stuff might be annoying.
2. What do you picture this interface looking like within hass? As far as I know there's no control for a switch with three options, I can easily write the code to change the mode (and retrieve the current one) but I might need some ideas for what the hass side of this should look like.

Thanks!

slackwilson, 2024-06-19:
Spam away!

1. Changing modes is the main thing I am after, and if that's all I can get I'll be a hell of a lot happier than I am now.
 
Changing the reserved%?, I'm not likely to change that much but it might be a useful method of trying to force the battery to charge or not charge by adjusting that to a high or low setting so if it's easy to add Reserved% then you might as well please. 

If it's the Hedge itself (on the app its simple on/off) you're referring to then again you may as well. 

I will admit it would be "nice to have" the ability to change TOU settings, but I can imagine there is a shedload to it, and really, once the TOU is configured within the app, changing things like block times of the different peak/off-peak/super off-peak etc is unlikely and not necessary to be able to do in HA. The only one I have been changing as a way to control how much I am charging (or not) is the "grid charge" option and only within the super-off peak time block. 
[image]

This is the one I am referring too, I dunno if you can just control that alone without having to delve into all the other options? 

2. Really not fussed how it looks within HA as long as it's changeable within an automation and I guess I can view the current setting.  HA looks to have drop down selection options within the gui - for example my Sensibo AC control device lets you select the AC's swing direction with a drop-down like this, maybe something like that?

[image]

npdsomerhayes, 2024-06-19:
Currently, I am changing between TOU and self-consumption each day.

I use the grid charge function within TOU to charge my aPower when I have free power form the network supplier and I use self-consumption the rest of the time.

If I use TOU while solar is being produced the system exports the solar instead of supplying the house load and charging the battery.  I do have advanced options in TOU that specifically tell it to supply the house, then the battery and then export but it does not work!

[image]

for my situation, it would be enough to change mode between Emergency Backup, Self-Consumption and Time of Use.  I would then use Home Assistant (actually node-red) to automate the changes.

[image]

However if I wanted to manually do it from home assistant I would most likely create a button for each mode . I use mushroom chip cards that can show state and have buttons. I think this would be easy enough to configure.

slackwilson, 2024-06-19:
@npdsomerhayes  I also change between Self-Consumption and TOU every day..  I have just signed up for a new energy plan that gives me 3 hours free energy between 11am-2pm, so I also swap to TOU in that time frame ( with Grid Charge on) to fully charge the battery (and heat my HWS etc) and then change to self consumption for all other times.

Like you, I also tried other settings and found they don't really work as expected. 

The only reason I would change that Grid Charge option.. is I have managed to trip my main circuit breaker drawing over 15kw from the grid, it only happened on a particularly dark cloudy day and my solar was dropping to 0, and I was pulling so much from the grid it tripped, changing that setting for a while let me avoid tripping the breaker - but I plan on getting a wifi timer/switch installed on the HWS so I can remotely turn that off in future if the same thing happens again.

richo, 2024-06-19:
For what it's worth if you put the HWS on a smart circuit in your aGate you'll be able to control it via home assistant :) That's how we control charging my gf's car only from excess solar production.

slackwilson, 2024-06-19:
Ah yer good idea, I need to check with my installer on that option. I suspect it will involve another cable run because the way our house is laid and and due to the regulations here, the agate ended up about 18 meters away from the main circuit panel.. meaning it may just be easier/cheaper to use a separate control. Using a FWH smart circuit is a good option if we can though.

richo, 2024-06-23:
I grabbed all the requests off the wire to knock this out, but wasn't able to get it working just yet. I'll let you know when I have something to test. Should be next couple of days I think.

richo, 2024-06-25:
https://github.com/richo/franklinwh-python/pull/1 Added the upstream code needed to change modes, I still need to figure out how to actually manifest this inside of home assistant.

slackwilson, 2024-06-25:
> [richo/franklinwh-python#1](https://github.com/richo/franklinwh-python/pull/1) Added the upstream code needed to change modes, I still need to figure out how to actually manifest this inside of home assistant.

Looking forward to it! Thanks for you efforts @richo

richo, 2024-06-25:
No worries! Happy to help and hopefully it's ok that I'm sometimes just brainstorming in these issues. I think I have a solution albeit a sort of ugly one. How would you feel about it if you did something like:

```yaml
select:
  - platform: franklin_wh
    username: "email@domain.com"
    password: !secret franklinwh_password
    id: 1005xxxxxxxxxxx
    modes:
      time_of_use: 20
      emergency_backup: 100```

This would let you change modes with a select entity in home assistant, but there wouldn't be a way for you to change those %'s later without editing the config and reloading.

slackwilson, 2024-06-25:
Just so I'm clear, the % is the "reserved energy for outage?"

If there was a way to also change the reserve % without having to edit the config I would prefer it to be honest.   

If it's easy to get mode change working now doing that way, and the ability to change % could potentially be added as another improvement later then that's fine, I'm more than happy to set that in config now and then potentially another upgrade to change the % as a later improvement?

richo, 2024-06-25:
It's mostly that I can't really find a way to express it in terms of home assistant entities.

Yeah, the % is that reserved energy for outage, so like for my home I would do something like:

```yaml
    modes:
      self_consumption: 20
      emergency_backup: 100```
```

Since we basically always run on Self Consumption mode but if we did want to flip to emergency backup for whatever reason, we would want it to kick over with 100% reserved. And then when we flipped back it would go to 20.

I also reached out to the homeassistant devs to see if they have any thoughts that I'm not seeing. I'm gunna see if I can get this working tonight and go from there. The other option is to expose it as a generic service, but then you would only be able to poke at it via the `call_service` primitive and doing things like putting the current mode on a dashboard would become much more involved.

richo, 2024-06-25:
ok I have things half working in https://github.com/richo/homeassistant-franklinwh/pull/13

It does update the soc properly but the numbers it returns aren't actually what it is set to. Setting the new values when you change modes does work correctly though.

slackwilson, 2024-06-26:
@richo I know you haven't said its ready to test yet, but I thought I'd try it anyway. I figured out how to get the mode-changing branch installed, I added the select into configuration.yaml with this option

select:
  - platform: franklin_wh
    username: "my@email"
    password: !secret franklinwh_password
    id: 100xxxxxxxxxxxxxxx
    modes:
      self_consumption: 5
      time_of_use: 5

After restarting I get the following error in the logs:

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:941
First occurred: 07:55:44 (731 occurrences)
Last logged: 13:58:46

Update for select.franklinwh_mode fails
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 941, in async_update_ha_state
    await self.async_device_update()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1291, in async_device_update
    await hass.async_add_executor_job(self.update)
  File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/franklin_wh/select.py", line 102, in update
    (name, soc) = self.client.get_mode()
                  ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/franklinwh/client.py", line 251, in get_mode
    mode_name = MODE_MAP[status["runingMode"]]
                ~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^

Not sure if that is a spelling mistake in that last one, has a single n in running instead of 2?

On the I guess good news front, it does indeed change the mode.. but the bad news is it doesn't matter what I choose it always changes the mode to "emergency backup" and it also seems to turn on the storm hedge option, so when I change the mode back to either "TOU" or "Self consumption" in the app, the storm hedge as on (I leave that off normally) which makes it want to charge from the grid until I turn that off.

Dunno if any of that is helpful or not :)

richo, 2024-06-26:
Hey, so the single n is intentional (that's what their API returns). It always setting the same mode and turning on storm hedge is not! Let me see if I can track down what's going on. This is super helpful, thankyou for sharing.

That error is in the codepath for getting the current mode from their API, not from updating the mode from your side. Was there anything below it? I would have expected something like a KeyError underneath. I'm wondering if you're getting something different back from the API for some reason.

richo, 2024-06-26:
Oh sorry, one other thing. Is your modes indented?

It should look like:

```yaml
- platform: franklin_wh
  username: "my@email"
  password: !secret franklinwh_password
  id: 100xxxxxxxxxxxxxxx
  modes:
    self_consumption: 5
    time_of_use: 5
```
not
```yaml
- platform: franklin_wh
  username: "my@email"
  password: !secret franklinwh_password
  id: 100xxxxxxxxxxxxxxx
  modes:
  self_consumption: 5
  time_of_use: 5
```

I'm guessing it's correct and it's just github eating your formatting but wanted to double check. I believe I did the config validation correctly so it'll yell at you if you have it wrong but I am figuring a great deal of this out as I go along :)

slackwilson, 2024-06-26:
yep just github that stripped the indentation when I pasted it, I have it indented in the cfg :)

Yup when I looked in the full log I could see a Key error - here is the last few repeated errors in the full log..

2024-06-26 16:57:18.111 ERROR (MainThread) [homeassistant.helpers.entity] Update for select.franklinwh_mode fails
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 941, in async_update_ha_state
await self.async_device_update()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1291, in async_device_update
await hass.async_add_executor_job(self.update)
File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/franklin_wh/select.py", line 102, in update
(name, soc) = self.client.get_mode()
^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/franklinwh/client.py", line 251, in get_mode
mode_name = MODE_MAP[status["runingMode"]]
~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
KeyError: 2
2024-06-26 16:57:48.200 ERROR (MainThread) [homeassistant.helpers.entity] Update for select.franklinwh_mode fails
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 941, in async_update_ha_state
await self.async_device_update()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1291, in async_device_update
await hass.async_add_executor_job(self.update)
File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/franklin_wh/select.py", line 102, in update
(name, soc) = self.client.get_mode()
^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/franklinwh/client.py", line 251, in get_mode
mode_name = MODE_MAP[status["runingMode"]]
~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
KeyError: 2
2024-06-26 16:58:18.154 ERROR (MainThread) [homeassistant.helpers.entity] Update for select.franklinwh_mode fails
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 941, in async_update_ha_state
await self.async_device_update()
File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1291, in async_device_update
await hass.async_add_executor_job(self.update)
File "/usr/local/lib/python3.12/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/franklin_wh/select.py", line 102, in update
(name, soc) = self.client.get_mode()
^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.12/site-packages/franklinwh/client.py", line 251, in get_mode
mode_name = MODE_MAP[status["runingMode"]]
~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^
KeyError: 2

richo, 2024-06-27:
Thanks for posting the complete error. It is.. very exciting that you get completely different values for runingMode than I do. For reference mine are `9322`, `9323`, and `9324`. And seemingly one of yours is just `2`.

Let me do a little more poking and see if there's a place I can interrogate to get what hte values all mean. That also probably explains why setting any value gives you emergency backup, I'm guessing `9324` (the value for emergency backup on mine) doesn't mean anything to yours.

Appreciate your help as always. I might commit a script that dumps some of this information from the API and ask you to run it for me, if that's ok. It would help me reversing it a lot.

slackwilson, 2024-06-27:
Happy to run anything you come up with

david2069, 2024-09-22:
> Thanks for posting the complete error. It is.. very exciting that you get completely different values for runingMode than I do. For reference mine are `9322`, `9323`, and `9324`. And seemingly one [...]

Hi Richo

(Been meaning to do this for ages - but only just finished)

I just had a poke around and observed the following:

It appears that the value of **runingMode** is the same as the value of **currentId**

The value of **currentId** is returned in the **result** of response after calling: "hes-gateway/terminal/tou/getGatewayTouList" 

This will return an array of the **oldIndex** values and the associated **id** values - one of which will be the same as **currentId**

The response **result** and array of the settings is found in **list** (id, oldIndex, name, soc, minSoc, minSoc, editSocFlag, etc)

For my aGate, currentId = 85232 and runingMode = 85232 (id: 85232, oldIndex: 2, workMode: 2, name: "Self-Consumption"
(the response from sendMqtt should match this).

_Example_ POST request body for updateTouMode 

`currendId=85232&gatewayId=10060006A02F24170091&lang=EN_US&oldIndex=2&soc=1&stromEn=1&workMode=2`

**touSendStatus** is where they store the status of the pending **workMode** change and the message is in touAlerMessage. It appears to clear after it is has been submitted (not if it is in affect on the aGate - but a call to getGetewayToList will confirm this).

Hence, you the value of **currentId** in the array **list** to look up the **oldInex** values, etc.

I note **todayVppVo** for VPP details (if your aGate is subscribed) - none here in Australia yet  - one day. But most seem like a bad idea these days! Not much incentive to use one.

I mention this, as no doubt a VPP will override these settings  - for the few cents or dollars they get for managing your batter. 

`{
  "code": 200,
  "message": "Operate success!",
  "result": {
    "currendId": 85232,
    "list": [
      {
        "id": 29287,
        "oldIndex": 3,
        "name": "Test Peak",
        "soc": 5.0,
        "maxSoc": 100.0,
        "minSoc": 0.0,
        "editSocFlag": true,
        "multiSOCFlag": false,
        "workMode": 1,
        "energyIncentivesType": null,
        "electricityType": 1,
        "displayFlag": null
      },
      {
        "id": 47522,
        "oldIndex": 1,
        "name": "Emergency Backup",
        "soc": 100.0,
        "maxSoc": 100.0,
        "minSoc": 0.0,
        "editSocFlag": false,
        "multiSOCFlag": false,
        "workMode": 3,
        "energyIncentivesType": null,
        "electricityType": 1,
        "displayFlag": null
      },
      {
        "id": 85232,
        "oldIndex": 2,
        "name": "Self-Consumption",
        "soc": 1.0,
        "maxSoc": 100.0,
        "minSoc": 0.0,
        "editSocFlag": true,
        "multiSOCFlag": false,
        "workMode": 2,
        "energyIncentivesType": null,
        "electricityType": 1,
        "displayFlag": null
      }
    ],
    "stromEn": 1,
    "sgipFlag": 1,
    "itcFlag": 1,
    "stopMode": 0,
    "gridChargeEn": 0,
    "touSendStatus": null,
    "touAlertMessage": null,
    "backupForeverFlag": 1,
    "timerEndTime": null,
    "timerEndTimeZero": null,
    "timerStartTime": null,
    "timerStartTimeZero": null,
    "zoneInfo": "Australia/Sydney",
    "nextWorkMode": null,
    "todayVppVo": {
      "programFlag": null,
      "vppStatus": null,
      "vppId": null,
      "vppFlag": 0,
      "startTime": null,
      "endTime": null
    }
  },
  "success": true`

Cheers,
dave