richo/homeassistant-franklinwh #10: Support generator status in sensors

richo, 2024-06-20:
The information appears to be in the mqtt info payload, shoudl just be a case of pulling it out and interpreting it.

npdsomerhayes, 2024-06-20:
I have this working! 
I was working on this yesterday as one of the main transmission towers feeding the region where I live "fell over"!
I had my generator running for about 4 hours yesterday so thought I would have a go at integrating it.  

I am a very amateur programmer and I ran into a couple of issues that I asked ChatGPT to help me with. ChatGPT changed the structure and I want to follow your structure so I will play with it today

richo, 2024-06-20:
That's great!

If you need some help, feel free to create a pr with what you have and I
can give you a hand getting it mergable :)

npdsomerhayes, 2024-06-20:
I think I have made the pr (I have not done one before) let me know if you get it if not my code is below

`"""Platform for sensor integration."""
from __future__ import annotations
from threading import Lock
import time

import franklinwh

import voluptuous as vol
import homeassistant.helpers.config_validation as cv

from homeassistant.components.sensor import (
    PLATFORM_SCHEMA as PARENT_PLATFORM_SCHEMA,
    SensorDeviceClass,
    SensorEntity,
    SensorStateClass,
)
from homeassistant.components.switch import SwitchEntity
from homeassistant.const import (
        UnitOfPower,
        UnitOfEnergy,
        PERCENTAGE,
        CONF_USERNAME,
        CONF_PASSWORD,
        CONF_ID,
        )

from homeassistant.core import HomeAssistant
from homeassistant.helpers.entity_platform import AddEntitiesCallback
from homeassistant.helpers.typing import ConfigType, DiscoveryInfoType
PLATFORM_SCHEMA = PARENT_PLATFORM_SCHEMA.extend(
        {
            vol.Required(CONF_USERNAME): cv.string,
            vol.Required(CONF_PASSWORD): cv.string,
            vol.Required(CONF_ID): cv.string,
            }
        )

def setup_platform(
    hass: HomeAssistant,
    config: ConfigType,
    add_entities: AddEntitiesCallback,
    discovery_info: DiscoveryInfoType | None = None
) -> None:
    """Set up the sensor platform."""
    username: str = config[CONF_USERNAME]
    password: str = config[CONF_PASSWORD]
    gateway: str = config[CONF_ID]

    fetcher = franklinwh.TokenFetcher(username, password)
    client = franklinwh.Client(fetcher, gateway)
    cache = CachingClient(client.get_stats)

    add_entities([
        FranklinBatterySensor(cache),
        HomeLoadSensor(cache),
        BatteryUseSensor(cache),
        GridUseSensor(cache),
        SolarProductionSensor(cache),
        BatteryChargeSensor(cache),
        BatteryDischargeSensor(cache),
        GeneratorUseSensor(cache),  # Adding the new generator sensor here
        ])  

UPDATE_INTERVAL = 60
class CachingClient(object):
    def __init__(self, update_func):
        self.mutex = Lock()
        self.update_func = update_func
[... 140 lines cut ...]

    def update(self) -> None:
        """Fetch new state data for the sensor.

        This is the only method that should fetch new data for Home Assistant.
        """
        stats = self._cache.fetch()
        self._attr_native_value = stats.totals.battery_discharge

class GeneratorUseSensor(SensorEntity):
    """Shows the current power output of the generator"""

    _attr_name = "FranklinWH Generator Use"
    _attr_native_unit_of_measurement = UnitOfPower.KILO_WATT
    _attr_device_class = SensorDeviceClass.POWER
    _attr_state_class = SensorStateClass.MEASUREMENT

    def __init__(self, cache):
        self._cache = cache

    def update(self) -> None:
        stats = self._cache.fetch()
        self._attr_native_value = stats.current.generator_production

`

npdsomerhayes, 2024-06-20:
I think I have created a pr (I have not done one before!) let me know if you don't get it and I will try to work out what I did wrong

richo-anyscale, 2024-06-20:
I see the branch on your fork, but you haven't created the PR yet. If you like, I can just go grab and pull it in but if you want the full experience, I think if you visit https://github.com/npdsomerhayes/homeassistant-franklinwh you should see a "Create pull request" banner somewhere near the top.

richo, 2024-06-20:
Closed by https://github.com/richo/homeassistant-franklinwh/commit/17b45a8314b70e6ac2133dba64e6deb1a230eee9

closed 2024-06-20 by richo