richo/homeassistant-franklinwh #14: Two FranklinWH installations at my house.

WaywardWilderness, 2024-07-18:
We have a permitted ADU with separate electrical service, and as a result we have 2 FranklinWH installs each with a battery and an A gate. I would like to see each of these in HA. Can I just have 2 sensors labeled something like this?
sensor:
  - platform: franklin_wh_house
    username: "email@domain.com"
    password: !secret franklinwh_password
    id: 1005xxxxxxxxxxx
  - platform: franklin_wh_apartment
    username: "email@domain.com"
    password: !secret franklinwh_password
    id: 1005xxxxxxxxxxx

richo, 2024-07-18:
I believe that should work, although now I stop to think about it, all of the entities name themselves without anything specific, so it might do something unexpected.

I can try to debug this idea with my house, but you won't break anything by trying it, so if you want to try and let me know what happens let me know. I'm not sure if you can assign entity IDs once they are created by the integration or if the integration needs to do it, but I think suffixing them with the gateway_id won't be hard, it will just make them unwieldy.

The last option would be to add a `nickname` optional key so that you can prefix/suffix the entities with it.

richo, 2024-07-18:
Thinking about it more I think just putting a nickname in so the entities inherit it is the right move (and this is how some of the other integrations I use work). Do you have strong feelings about how you'd expect that to work? if not I will sketch something out and add it.

WaywardWilderness, 2024-07-19:
I just installed HA today, so anything you suggest is fine with me. HA found a bunch of stuff on our network, but my very first order of business was to get your work installed. I'm particularly interested in integrating my e-gauge devices and the FranklinWH units so we can have a easy to understand display of what is going on with our power usage.

richo, 2024-07-19:
Sweet! Feel free to reach out if I can help. This is a passion project but I'm glad people are getting value out of it.

Getting it into HACS is a bit of a chore (mostly because the way they package it makes it annoying to install out of HACS) but I will be doing that once it's a little more stable that should make it easier for folks to setup. I'll update this issue once I have this in place for you.

richo, 2024-07-19:
I just pushed a branch called `nicknames` that I think should work, but I haven't tested it yet. I can when I get home.

richo, 2024-11-27:
@WaywardWilderness on the offchance, did you ever get a chance to test this? I think from memory I had some thoughts about why it wouldn't work, I can try to test it when I'm back at home for a day or two.

WaywardWilderness, 2024-11-27:
richo,

I have HA running in docker and I hadn’t spent much time with it. I failed to realize that I can’t install add-ons when I’m not running HAOS. At some point I plan to try HAOS on an Intel NUC I’ve got lying around. But I haven’t gotten that done and right now I am just not working with HA much. 

Richard

richo, 2024-11-27:
Ah no worries. I kinda suspect you can install them within HA in docker but doing so might be annoying.

I'm gunna shelve this for a while until someone else needs it but thanks for the idea!

WaywardWilderness, 2024-11-28:
Sorry to not have told you awhile back that I wasn’t actively working on my HA setup. That was rude of me and I’m sorry to have wasted your time. 

Richard

richo, 2024-11-28:
No need to apologise. This is a passion project, I had been very unmotivated for a few months and I'm just catching up now and trying to figure out where things are at. I appreciate you opening the issue in the first place, I hadn't really considered two aGates in one install.