richo/homeassistant-franklinwh #15: Grid outage sensor/identification

slackwilson, 2024-07-23:
Would be awesome if there was a sensor or some way of detecting when the grid is down?

I plan on setting up a bunch of automations, like turning on HVAC at certain times of the day if there is plenty of battery charge left, but I would not want the automations to run if there was a grid outage.

Not sure if there is anything obvious in the API you can use, but be great if there is and we can have that in home assistant :)

npdsomerhayes, 2024-07-24:
If this is not in the API I had already been thinking about doing this with a Shelly input module, powering it from the backed up loads and have the input from the non-backup loads Home assistant will see the contact open when the mains power is off.
I have not tested this yet but I don't see why it should not work. Just make sure you don't create a link between the backed up and non backup loads you could use an additional relay if you had any doubt about the isolation between the two.

slackwilson, 2024-07-24:
Cool idea but not sure I could do that with my setup very easily.  I don't have any non-backed up loads, everything is on backup/battery. I guess I could get an electrician to wire up an outlet on the non-backup side, but of course that would cost me a call-out. I'll think about what else I can do, but hopefully the API has something usable.

richo, 2024-07-24:
I'm pretty confident it does- the app knows when the grid is out.

david2069, 2024-09-04:
> I'm pretty confident it does- the app knows when the grid is out.

Hi Richo

Great work ! I’m enjoying using it…bern over a month now….

RE: Grid status
I observed that the response to POST
https://energy.franklinwh.com/hes-gateway/terminal/sendMqtt

when off grid adds it adds a new key/value pair:
`offgridreason:1`

When the grid is connected it is not present.

Other API calls return this value as well. 

Although, they recently changed getDeviceInfo and no longer does.  It a new array structure: peHwVerList (lists your installed aPowers ) and seems to always sets offGridReason to null…likely a bug?

I also observed that app polls the off grid status:

https://energy.franklinwh.com/hes-gateway/terminal/getOffGridFlag?gatewayId=xxxxxx1&lang=EN_US

offGridFlag: 0

 It appears when you request to go back to the grid (clicking on “Reconnect”) sends this request:

POST https://energy.franklinwh.com/hes-gateway/terminal/updateOffgrid

{
  "gatewayId": "xxxxxxxxxxc",
  "lang": "EN_US",
  "offgridSet": 0,
  "offgridSoc": 5
}

Cheers,
David

richo, 2024-09-07:
Hey David,

Thanks so much for this. It's really helpful. Sorry I haven't had much time to think about this lately.

I think my plan for the short term probably is something along these lines:

- Treat the current feature set as being stable, and work on getting it into HACS. I got an email from someone who was struggling to install it, and who I unfortunately didn't hear back from, but I think I would like to streamline the "just works" experience a little bit.

- Once that's done, I think I want to change the way the sensors work from all polling upstream and trying to make sense of replies, to having one central process that is polling their API and publishing events to an internal message bus. Given that there's a few ways to figure out whether or not it's attached to the grid, I think I'd rather do it right.

Buuuut if folks are hurting for not having this functionality I could be convinced to get something working for this, then do the HACS stuff, and then tackle that refactor.

richo

david2069, 2024-09-09:
Hey Richo 

Sounds like good idea to leave current release as is until you get a HACS release done.

FYI; 
I had mild panic when one of my HA automations informed my FHP SoC was zero %!

I checked the FranklinWH and it annoying was going  an upgrade with no notice or event or notification it was or has done it.

They did a fix/upgrade to aGate upon on installation day (22 July) to get  my Enphase Microinverters IQ7A to work. As it not work when first connected.

Looking for changes I can see so far:

https://energy.franklinwh.com/hes-gateway/terminal/getHomeGatewayList?lang=EN_US

Post upgrade now returns (after Quick Look):

       "protocolVer": "V1.11.01",
      "version": "V12R01B01D05_240902"

previously it was:

      "protocolVer": "V1.11.00",
      "version": "V12R01B22D00_240724"

Of the keys I have checked for FHP aGate hardware are they same - what are yours? Mine  is a late July installation here in Sydney.

     "realSysHdVersion": "FHP1.2",
      "sysHdVersion": "102"

I have the two Smart Circuits installed too. I had to remove the three switch setup in your default config to get to work properly.

Cheers 
David

david2069, 2024-09-22:
Hi Richo

While logging at the logs - here are my observations when the aGate goes offgrid and I have an active app connection:

a) The **sendMqtt** response result has the following added key/value pair: **offgridreason: 0**. otherwise, it is not present whilst on grid.

b) if the response result of a call has the key/value pair  **offgridreason: 0**. Whilst on grid this value is: **offgridreason: -1**

c) The app polls for Gateway alarms frequently and receives response whilst the alarm related to the off-grid (or other error I imagine) event exists. An alarm event is returned and repeats over and over to the app but is not shown to the end user. Probably an installer login it does, but that’s speculation.

The following polling occurs and it returns **alarmCode** 3001 initially and later 3007 is added. Only for a true not the app simulated outage. 

Note that when you login a complete list of the alarmCode numbers and short description is downloaded (257 currently) into the app  (.../common/selectDeviceRunLostList?lang=EN_US).

[https://energy.franklinwh.com/hes-gateway/terminal/selectGatewayAlarm?gatewayId=XXXXXXXXXXXXXXXXXXXXX&lang=EN_US](https://energy.franklinwh.com/hes-gateway/terminal/selectGatewayAlarm?gatewayId=10060006A02F24170091&lang=EN_US)

`{
  "code": 200,
  "message": "Query success!",
  "result": [
    {
      "id": 164472024,
      "gatewayId": "removed",
      "alarmEqSn": "removed",
      "alarmCode": "3001",
      "commonAlarmCode": "3001",
      "level": 2,
      "time": "2024-09-21 16:23:00",
      "logName": "Grid Abnormal",
      "commonLogName": "市电故障",
      "logNameEn": "Grid Abnormal",
      "commonLogNameEn": "Grid Abnormal",
      "createTime": null,
      "alarmExplanationZh": "Abnormal mains voltage or frequency",
      "alarmExplanationEn": "① Caused by poor power grid quality",
      "alarmExplanation": "① Caused by poor power grid quality",
      "pictureUrl": null,
      "plan": null
    },
    {
      "id": 164472025,
      "gatewayId": "removed",
      "alarmEqSn": "removed",
      "alarmCode": "3007",
      "commonAlarmCode": "3007",
      "level": 2,
      "time": "2024-09-21 16:23:10",
      "logName": "Grid Level 1 Undervoltage Protection",
      "commonLogName": "市电一级欠压保护",
      "logNameEn": "Grid Level 1 Undervoltage Protection",
      "commonLogNameEn": "Grid Level 1 Undervoltage Protection",
      "createTime": null,
      "alarmExplanationZh": "The voltage is lower than the first-level undervoltage protection value",
      "alarmExplanationEn": "①Due to poor power grid quality, CA mains voltage is <105.6V\n② The neighbor's load is short-circuited and the mains switch cannot be turned off.\n③The electric meter is not calibrated before leaving the factory, and the voltage value collection is inaccurate.",
      "alarmExplanation": "①Due to poor power grid quality, CA mains voltage is <105.6V\n② The neighbor's load is short-circuited and the mains switch cannot be turned off.\n③The electric meter is not calibrated before leaving the factory, and the voltage value collection is inaccurate.",
      "pictureUrl": null,
      "plan": null
    }
  ],
  "total": 2,
  "success": true
}`

Cheers
dave

jkt628, 2025-06-02:
this would be really useful.  the biggest consumers of electricity in my installation are EVs, by far!  both [Tesla](https://www.home-assistant.io/integrations/tesla_fleet) and [Rivian](https://github.com/bretterer/home-assistant-rivian) integrations offer switches to stop charging so i can avoid wiping out batteries if the grid fails without requiring smart switch.

jkt628, 2026-01-11:
i have it in @JoshuaSeidel fork and started moving here in https://github.com/richo/homeassistant-franklinwh/pull/60.  @richo and i have several balls in the air and chat regularly.