richo/homeassistant-franklinwh #17: Local Access?

Arrgg, 2024-10-22:
Is there any hope of local access to the aPower, via ethernet or serial?

I would love to get full access to the service port data so I can be more selective about how I log data and integrate into my InfluxDB system for graphing and logging.

richo, 2024-11-06:
heya. Short answer, yes it appears so.

Longer answer: I haven't made any real attempt at reversing it. There's a AP specific hotspot that exposes more functionality but shoving a sniffer in between is kinda hassle and I haven't invested that time yet.

npds188, 2026-10-15:
For anyone following this: local access works. I've released a Home Assistant integration (built with Claude.ai and Codex) that talks to the aGate directly on the LAN, with no cloud needed:

https://github.com/npds188/homeassistant-franklinwh-local

Direct Connect (TCP 9000) the interface the FranklinWH app uses over its hotspot. It's reachable on the normal LAN too, polls every 5 s and covers power flow, battery cells, generator, smart circuits and mode control.

SunSpec Modbus TCP (port 502) as @sdwilsh mentioned: read-only, and it has to be enabled. In AU/NZ, FranklinWH support enabled it for me once I explained it was for Home Assistant.

 I’ve also offered Rico in #89 to fold local access into this integration.

david2069, 2026-10-15:
> For anyone following this: local access works. I've released a Home Assistant integration (built with Claude.ai and Codex) that talks to the aGate directly on the LAN, with no cloud needed: [...]

I take except to this code as you have obviously cloned it with taking the sufficient credit to my work:

https://github.com/david2069/franklinwh-direct-connect-api

npds188, 2026-10-15:
> 5\. > For anyone following this: local access works. I've released a Home Assistant integration (built with Claude.ai and Codex) that talks to the aGate directly on the LAN, with no cloud needed:
>     > https://github.com/npds188/homeassistant-franklinwh-local
>     > Direct Connect (TCP 9000) the interface the FranklinWH app uses over its hotspot. It's reachable on the normal LAN too, polls every 5 s and covers power flow, battery cells, generator, smart circuits and mode control.
>     > SunSpec Modbus TCP (port 502) as [@sdwilsh](https://github.com/sdwilsh) mentioned: read-only, and it has to be enabled. In AU/NZ, FranklinWH support enabled it for me once I explained it was for Home Assistant.
>     > I’ve also offered Rico in [#89](https://github.com/richo/homeassistant-franklinwh/issues/89) to fold local access into this integration.
>     
>     
>     I take except to this code as you have obviously cloned it with taking the sufficient credit to my work:
>     https://github.com/david2069/franklinwh-direct-connect-api

Hi David, thanks for raising this. I want to get it right.

My aim was a full Home Assistant integration (UI setup, coordinators, controls, Modbus backup and so on) rather than a library, which is why I started a new repo instead of forking yours. Your work was a real foundation for it, and I credited you from the first commit, at the top of the README and in the Credits section, for the local command catalog, the battery reads and the mode and generator write recipes:

That said, you have a fair point: two functions (generator settings and smart-circuit switching) were adapted from your code, and under your MIT licence your copyright notice should be included, not only a credit line. I’m happy to add your notice and licence, mark the adapted code, and make the README credit clearer, with links to your library, PyPI and docs. I should also have mentioned you in my post on #17. Sorry about that.

Would that address your concerns, or is there anything else you’d like credited or changed?

david2069, 2026-10-16:
I have given up with this repo and it not going any where .... so done my own... first the Cloud API, Modbus API (new async version coming), Direct Connect API (has a lot more functionality than yours and is on PyPi), etc... and new FranklinWH Gateway emulator.

There is already available (freshly released) local integrations for FranklinWH aGates and aPower S - these area early Alpha-level releases of **both** Modbus TCP and Direct Connect Bridge integration which is compatible with Home Assistant App (Addon) or can be run standalone or under Docker or other virtualisation environments without Home Assistant:

To overcome limitations of HA the have optional UI / dashboards - you do not need to use them. They have advanced and basic functions, Advanced functions include Home Assistant integration to HA Entities via Websockets, Open Meteo Weather API built-in, Schedulers with optional conditions (Built-in and HA Entities), electricity utility billing tariff setup and tracking, etc.

You can use the native REST-API or optional MQTT via HA Discovery protocol for control, sensors and diagnostics. 
https://github.com/david2069/franklinwh-direct-connect-bridge/blob/main/INSTALL.md
Home Assistant add-on

[image]

Click the button above, or add the repository by hand:
1. In Home Assistant: **Settings → Add-ons → Add-on Store**.
2. **⋮** (top right) → **Repositories**, add
   `https://github.com/david2069/franklinwh-direct-connect-bridge`, then close.
3. Find **FranklinWH Direct Connect Bridge** and click **Install**.
4. On **Configuration**, set `fwh_host` to the aGate's IP. Set `mqtt_enabled: true` to
   publish the battery as Home Assistant entities. **Save.**
5. **Start**, then open the sidebar panel (served over HA ingress).

[image]

**MODBUS BRIDGE**
https://github.com/david2069/franklinwh-modbus-bridge/blob/main/INSTALL.md
Home Assistant add-on

[image]

Click the button above, or add the repository by hand:

1. In Home Assistant go to **Settings → Apps → App store** (called
   **Add-ons → Add-on Store** in older versions).
2. Open **⋮ → Repositories**, paste
   `https://github.com/david2069/franklinwh-modbus-bridge` and click **Add**.
3. Find **FranklinWH Modbus Bridge** in the store and click **Install**. The
   first install builds the image on your HA host and compiles a few Python
   wheels, so it takes several minutes (longer on a Raspberry Pi).
4. On the **Configuration** tab set:

[image]

david2069, 2026-10-16:
> 6\. > 5. > For anyone following this: local access works. I've released a Home Assistant integration (built with Claude.ai and Codex) that talks to the aGate directly on the LAN, with no cloud needed:
>     > > https://github.com/npds188/homeassistant-franklinwh-local
>     > > Direct Connect (TCP 9000) the interface the FranklinWH app uses over its hotspot. It's reachable on the normal LAN too, polls every 5 s and covers power flow, battery cells, generator, smart circuits and mode control.
>     > > SunSpec Modbus TCP (port 502) as [@sdwilsh](https://github.com/sdwilsh) mentioned: read-only, and it has to be enabled. In AU/NZ, FranklinWH support enabled it for me once I explained it was for Home Assistant.
>     > > I’ve also offered Rico in [#89](https://github.com/richo/homeassistant-franklinwh/issues/89) to fold local access into this integration.
>     > ```
>     > I take except to this code as you have obviously cloned it with taking the sufficient credit to my work:
>     > https://github.com/david2069/franklinwh-direct-connect-api
>     > ```
>     
>     
>     Hi David, thanks for raising this. I want to get it right.
>     My aim was a full Home Assistant integration (UI setup, coordinators, controls, Modbus backup and so on) rather than a library, which is why I started a new repo instead of forking yours. Your work was a real foundation for it, and I credited you from the first commit, at the top of the README and in the Credits section, for the local command catalog, the battery reads and the mode and generator write recipes:
>     That said, you have a fair point: two functions (generator settings and smart-circuit switching) were adapted from your code, and under your MIT licence your copyright notice should be included, not only a credit line. I’m happy to add your notice and licence, mark the adapted code, and make the README credit clearer, with links to your library, PyPI and docs. I should also have mentioned you in my post on [#17](https://github.com/richo/homeassistant-franklinwh/issues/17). Sorry about that.
>     Would that address your concerns, or is there anything else you’d like credited or changed?

Your has the basics, but if you are waiting for it be added here - it is not the right place - in my view.