richo/homeassistant-franklinwh #19: Project status?

riddley, 2024-11-27:
I came across this project a couple of months ago, but wasn't able to get it working. Is this project still being developed? Would a monetary contribution help to move things along? Thanks for making this!

richo, 2024-11-27:
Hey Riddley,

Yes I'm still working on it, although it's not my first priority. I appreciate the offer of a monetary contribution, but that's not the issue. If you get it working, and like it, I would be happy to include some charities people could contribute to instead in the README.

What issues are you having getting it working? I'd be happy to see if I can help.

richo

riddley, 2024-11-27:
It was more that I bounced off of it as I didn't want to make manual changes to my HA setup. Perhaps I'm fooling myself that HACS is doing a good job of cleaning up when/if I uninstall something, but as a long time Linux user I've learned that making manual changes often creates regret.

richo, 2024-11-27:
Ah I am with you. I had been putting off putting it up on HACS for a while now, I have a few features that aren't finished and I think I let perfect be the enemy of good.

I am fine with moving it to HACS tonight if that would help, I can still get my features in there but I think the upgrade path for folks who are currently using a branch might be more complicated. I think on balance that's an ok tradeoff.

riddley, 2024-11-27:
Wow that would be amazing. I don't mean to commandeer your night though. Please release it on HACS when you're ready and just know that I'll be excited about it when you do. Thanks again!

richo, 2024-11-27:
No you're fine. Honestly, hearing from people who are using it/want to use it is far more motivating to me than the money would have been. I have a few things to crack onto, but I think you've convinced me to just get it up on HACS and go from there.

I'll tag you in this issue when it's up.

jkt628, 2025-01-13:
today i'm getting a new system installed and looked around for an integration.  i don't see this published on HACS.  @richo would you mind updating the current status here?  TIA :{)}

richo, 2025-01-13:
Hey @jkt628 

Yeah, so I've been working on it but ran into some bugs upstream, I have a circularish dependency on https://github.com/hacs/integration/pull/4230 or something like it going in before I really want to do it in this repo, mostly because I don't want to cause more chaos than I have to in the migration.

mikeyroth, 2025-04-03:
Any update on HACS?   I really want to use this to make an automation for my pool heater.

richo, 2025-04-08:
hey, yeah I'm still looking at it. I'll be honest I'm frustrated that there's not an obvious way to support hacs without making the experience of setting it up for non HACS folks quite a bit more annoying. Maybe I'm just being unrealistic and people will just want to use HACS anyway.

richo, 2025-05-04:
hey just to keep everyone in the loop, since I know this is still a priority. I have a [branch](https://github.com/richo/homeassistant-franklinwh/tree/caching-client) I am testing at home which makes updating much more reliable and appears to have fixed that crash that happens occasionally when the gateway times out a lot.

I'm going to migrate that to the smart switch stuff as well, figure out how to ship mode changing, and then I'm gunna bite the bullet and publish on hacs.

slackwilson, 2025-05-06:
@richo I've been quiet on this, I haven't been hassling as I know you wanted to focus on publishing via hacs.. anyway I think I was one of the main requestors for mode changing, and I made that request based on an old version of the WH app which was pretty limited.

While I was waiting I ended up using an old android phone and macrodroid to change modes at certain times of day to get around the inability to do what I wanted either in the app, or via your add-on using HA.

However now with the latest version WH app, I have been able to schedule the ability to effectively do what I wanted based on time of day directly in the WH app. This has pretty much stopped the need for me to do mode changes.. now that requirement for me has basically reduced right down.  I'm happy for the request to drop right down in your priority or even be taken off altogether, although there is probably others that might find it useful, and it does give more options for automations if it was available :)