richo/homeassistant-franklinwh #22: Entities device_class not Energy?

jhuettner, 2024-11-29:
Thanks so much for creating this!  Us Franklin users would be out in the cold without you!  I'm trying to add my Franklin system to Home Assistant's Energy screen.  However it won't allow me to select the Franklin Entities because it says their device_class is not "energy".  Directs me to https://www.home-assistant.io/docs/energy/faq/#troubleshooting-missing-entities.  Can we add that to these entities?

jhuettner, 2024-11-29:
Wow, you're doing this for free, don't worry about setting expectations!  Enjoy the trip!  Wish I knew as much as you so I could help!
   I think it's updating the class of the entities.

I'm trying to set up the Energy page:

[image]

I try to add consumption on the Electricity Grid:

[image]

But it won't allow me to pick an entity for the consumed energy.  I think it's because the sensor.franklinwh_grid_use entity is does not have the device_class set to energy.  I also can't edit it from the entities screen because Home Assistant says they don't have unique id's.

[image]

   Appreciate all the help and for creating the integration!

richo, 2024-11-29:
Yeah absolutely. So the grid_use entity I'm fairly sure is Kw not KWh (off the top of my head, would have to go look) which is probably why it won't let you select it. It's indicating power, not energy. That said I'm fairly sure that there's an energy value in the payload they return to us, and if not aggregating it wouldn't be that hard.

I'll update this issue when I get to have a look. And yeah, I don't feel bad taking a break, but it's also exciting to me that other folks are actually using this and so I want to make sure people know the project isn't dead when I don't respond for a week :)

bradyr, 2024-11-29:
@jhuettner heya, i've been watching these posts -- i think i can help out with this one.   you can create some helpers in home assistant to massage the data from the franklin integration in order to dump into the HA energy dashboard.  

here's some quick and dirty info on the helpers that i created to accomplish this task (please note i havent updated my integration to the latest hacs version -- i've been letting it settle down since i know @richo just did the migration and my current config using the "old" code is still rock solid)

`
example template to change the battery current charge/discharge to watts:

name: FranklinWH - Battery - current charge (watts) (or discharge)

unit of measurement: W 
device class: power
state class: measurement

charging:
{{ max(states.sensor.franklinwh_battery_use_watts.state | int, 0) | default(0)}}

discharging:
{{ min(states.sensor.franklinwh_battery_use_watts.state | int, 0) | abs | default(0)}}

example integral sensor settings for changing the current charge/discharge to kWh:

name: FranklinWH - Battery - energy usage - charging (kWh)

unit of measurement: kWh

input sensor: FranklinWH - Battery - current charge (watts) (<-- this is one of the helpers created to convert the current charge/discharge to watts)

integral sensor options:
Left Riemann sum, precision = 2
`

i'm sure i'm missing some basic info but i just woke up so hopefully this will help out.  feel free reach out for more info.

jhuettner, 2024-11-30:
Oh, that's a good idea.  I'm still a Home Assistant newbie and I never thought of that.  Helpers are still pretty new to me.  Thanks for the tip!

david2069, 2024-12-14:
> Hey yeah of course! I'm glad it's useful to people.
> 
> Yes, I absolutely can. I might need a sec to find out if what I'm meant to be doing here is adding more entities or updating the classes of the existing ones that I have. I think it's the former though.
> 
> Can you describe exactly the behaviour that you're expecting here, to make sure we are on the same page? Eg, are you trying to track instantaneous use, or the aggregate over the course of some time period?
> 
> And finally, just to set expectations I'm leaving on a trip for a week tomorrow so you probably won't actually see an update from me until the week of the 8th.
> 
> Thankyou for the request!

You could simply look and reuse method used in  the official Home Assistant battery integration code - like the one for Powerwall 2:
https://github.com/home-assistant/core/blob/dev/homeassistant/components/powerwall/sensor.py
😁

richo, 2025-02-17:
Thanks for the reminder. Having a look at this today. I think it should be easy enough to add I just haven't gotten around to it.

richo, 2025-02-27:
Ok so I finally got around to doing this and it.. looks like maybe I already did?

https://github.com/richo/homeassistant-franklinwh/blob/main/sensor.py#L181-L217

Do these sensors work for you? If so it's very easy to add them as well for grid use/solar production etc.

jhuettner, 2025-03-03:
I think you would have to add additional sensors for the energy usage.  Right now it's only showing the instantaneous power, and HA's Energy page requires sensors that are of type Energy (like the ones you have for battery usage) but for the grid.

richo, 2025-03-03:
ok sweet. Yeah sorry that's what I was asking, but maybe in a really
oblique way. If the energy sensors for the battery work, adding more of
them for the grid is really easy.

richo, 2025-03-03:
Ok so I added all the sensors in that PR, but for whatever reason rebooting my hass hasn't added them as entities? I'm not really sure what I've done wrong but I don't want to merge it until I figure that out. With that said, the old Battery Charge and Battery Discharge sensors (from which I copied the new ones) have data that makes sense so I think this should Just Work once I figure that out.

richo, 2025-03-03:
Ok last update, I did something real stupid. Fixed it, but I'm not going to merge this until tomorrow since I'm clearly not firing on all cylinders. Thanks for continuing to hassle me about this, sorry it took so long.

PR #26 (2025-03-13): Grid energy sensors
Fixes #22
closed 2025-03-13 by richo