richo/homeassistant-franklinwh #23: Issue setting up my FranklinWH on HA

thamindzzeye, 2025-02-06:
The nontype seems to be failing successfully login and get an access token. But I have provided my email and password in the config. Any help would be greatly appreciated!!!

Here are my basic logs:

Logger: homeassistant.components.switch
Source: helpers/entity_platform.py:366
integration: Switch ([documentation](https://www.home-assistant.io/integrations/switch), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+switch%22))
First occurred: 2:04:21 PM (2 occurrences)
Last logged: 2:04:21 PM
Error while setting up franklin_wh platform for switch: 'NoneType' object is not subscriptable

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 366, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/franklin_wh/switch.py", line 47, in setup_platform
    client = franklinwh.Client(fetcher, gateway)
  File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 150, in __init__
    self.refresh_token()
    ~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 173, in refresh_token
    self.token = self.fetcher.get_token()
                 ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 112, in get_token
    return TokenFetcher.login(self.username, self.password)
           ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 133, in login
    return json['result']['token']
           ~~~~~~~~~~~~~~^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable

And the more advanced. 
33m2025-02-03 22:51:45.066 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration franklin_wh which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant[0m
[33m2025-02-03 22:51:45.067 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant[0m
[31m2025-02-03 22:51:47.279 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up franklin_wh platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 366, in _async_setup_platform
    await asyncio.shield(awaitable)
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/franklin_wh/sensor.py", line 51, in setup_platform
    client = franklinwh.Client(fetcher, gateway)
  File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 150, in __init__
    self.refresh_token()
    ~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 173, in refresh_token
    self.token = self.fetcher.get_token()
                 ~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 112, in get_token
    return TokenFetcher.login(self.username, self.password)
           ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 133, in login
    return json['result']['token']

richo, 2025-02-10:
Hi,

Sorry I didn't get back to you sooner. You're right, that does seem to be failing to login. I can push an update with slightly more verbose logs that will help today, but first thing would be to double check that you definitely have the correct username and password?

Are you using the home assistant secret store, or just putting them directly in the config?

thamindzzeye, 2025-02-11:
Hi Richo,

First of all, THANK YOU FOR THIS... I have a bit of a complex install, with SPAN and solar, 5 aPowers etc.. .

but yes I tried with the password in the secret yaml and also just directly in config.

since my password and id both contain letters and numbers should they too be in quotes in the config? like this 

sensor:
  - platform: franklin_wh
    username: "email@mail.com"
    password: "somepassword123"
    id: "12345A12W12345"

which items need quotes which dont? I'm pretty new to HA. 

my SN for my aGate starts in 1013 ... I had the brain updated when I got my franklin + SPAN.

I'm a YouTuber (youtube.com/twobitdavinci) and a test bed for FranklinWH I have a lot of new products before they ship, happy to chat if you ever want to!

richo, 2025-02-11:
With our without quotes shouldn't make too much difference, it's not a home assistant thing so much as a yaml thing, although yaml is a very exciting and neednessly complex file format. It's generally safer to just quote everything though if you're not sure.

Sorry last couple of days have been a bit chaotic (and I've been putting off a fairly substantial rewrite which will make things more reliable). I'll try to get that branch up for you today.

From what I've seen though, that error should really only happen if the username or password is incorrect. I can't imagine it's something specific to your setup, but I guess it's possible. Out of curiosity, what geographic region are you in? 

Also, when you go to the App can you get the Protocol Version and Hardware Version for your aGate? I don't really see how this can be it (my homeassistant integration is masquerading as the app, not the aGate so it shouldn't really matter but trying to narrow things down).

I have protocol version v1.11.00 and hardware 1.0 here.

thamindzzeye, 2025-02-11:
Protocol Version: V1.11.01
Hardware Version: 1.0

I'm in San Diego California

richo, 2025-02-11:
Ok so I went and refreshed my memory, and the code notionally handles an invalid password (With a different error) so something else goofy is happening. Just for completeness sake, can you try a wrong password and grab the logs from that?

Separately, I'm not sure what your comfort level is with this stuff, the python library that bridges homeassistant to franklin is here: https://github.com/richo/franklinwh-python

There's a script in the `bin` folder that should just fetch a token. Are you comfortable trying to run it and see what happens? If you're able to do that, I can easily add some instrumentation to it on a branch and we should be able to fix whatevers broken pretty quick at that point.

thamindzzeye, 2025-02-14:
I'm familiar with python, just need steps to run this individually.

I've cloned the project locally. What would be the next steps to run this, from terminal?

richo, 2025-02-15:
ok great!

So once you've cloned `franklinwh-python` I'll need you to check out the `debug-logging` branch. And then with a python environment ready you should be able to do something like:

```bash
virtualenv env
. env/bin/activate
pip install -r requirements.txt
python bin/login.py [username] [password]
```

It should spit out a line something like `INFO:franklinwh.client:JSON payload: {'code': 400, 'message': 'Account or password error!', 'result': None, 'success': False}` that I'm hoping will shine some light on what's not working for you. Thanks! let me know if you have issues.

richo

riddley, 2025-02-16:
I'm attempting this too. Do I need to do all of the steps on my HomeAssistant machine? I was trying to do the login.py on my daily driver linux box and got the following:

```
$ git clone https://github.com/richo/franklinwh-python.git                                              
Cloning into 'franklinwh-python'...                                                                                         
remote: Enumerating objects: 194, done.                                                                                     
remote: Counting objects: 100% (194/194), done.                                                                             
remote: Compressing objects: 100% (121/121), done.                                                                          
remote: Total 194 (delta 113), reused 147 (delta 67), pack-reused 0 (from 0)                                                
Receiving objects: 100% (194/194), 26.58 KiB | 3.32 MiB/s, done.                                                            
Resolving deltas: 100% (113/113), done. 

$ cd franklinwh-python/ && virtualenv env                                                                    
created virtual environment CPython3.12.9.final.0-64 in 50ms                                                                
  creator CPython3Posix(dest=/home/sam/src/franklinwh-python/env, clear=False, no_vcs_ignore=False, global=False)           
  seeder FromAppData(download=False, pip=bundle, via=copy, app_data_dir=/home/sam/.local/share/virtualenv)                  
    added seed packages: pip==25.0                                                                                          
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator   

$ . env/bin/activate

$ git switch debug-logging                                                          
branch 'debug-logging' set up to track 'origin/debug-logging'.                                                              
Switched to a new branch 'debug-logging'        

$ pip install -r requirements.txt                                                   
Collecting certifi==2024.2.2 (from -r requirements.txt (line 1))                                                            
  Using cached certifi-2024.2.2-py3-none-any.whl.metadata (2.2 kB)                                                          
Collecting charset-normalizer==3.3.2 (from -r requirements.txt (line 2))                                                    
  Using cached charset_normalizer-3.3.2-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (33 kB)         
Collecting idna==3.6 (from -r requirements.txt (line 3))                                                                    
  Using cached idna-3.6-py3-none-any.whl.metadata (9.9 kB)                                                                  
Collecting requests==2.31.0 (from -r requirements.txt (line 4))                                                             
  Using cached requests-2.31.0-py3-none-any.whl.metadata (4.6 kB)                                                           
Collecting urllib3==2.2.1 (from -r requirements.txt (line 5))                                                               
  Using cached urllib3-2.2.1-py3-none-any.whl.metadata (6.4 kB)                                                             
Using cached certifi-2024.2.2-py3-none-any.whl (163 kB)                                                                     
Using cached charset_normalizer-3.3.2-cp312-cp312-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (141 kB)                   
Using cached idna-3.6-py3-none-any.whl (61 kB)                                                                              
Using cached requests-2.31.0-py3-none-any.whl (62 kB)                                                                       
Using cached urllib3-2.2.1-py3-none-any.whl (121 kB)                                                                        
Installing collected packages: urllib3, idna, charset-normalizer, certifi, requests                                         
Successfully installed certifi-2024.2.2 charset-normalizer-3.3.2 idna-3.6 requests-2.31.0 urllib3-2.2.1           

$ python bin/login.py welcome@zombo.com WELCOME
Traceback (most recent call last):                                                                                          
  File "/home/zombo/src/franklinwh-python/bin/login.py", line 7, in <module>                                                  
    from franklinwh import TokenFetcher                                                                                     
ModuleNotFoundError: No module named 'franklinwh'
```

If this needs to be done on the home-assistant box, I'll delete this comment.

richo, 2025-02-17:
Hey no you're good, I think I missed a step in the instructions I gave you.

If you do a `pip install -e .` it should add that module to the virtualenv as well. If you start a new shell, you'll need to do `. env/bin/activate` again to load the virtualenv into your environment. Thanks for getting back to me and sorry for giving you bad instructions, I wasn't actually near my computer.

richo, 2025-02-17:
Oh sorry, and to answer your question, no it doesn't need to be your homeassistant machine.

riddley, 2025-02-17:
I'm much closer! BTW, I'm not the OP just in case my hijack was confusing.

When I run login.py I get some output that says "Your token is" and the token starts with "APP_ACCOUNT:" In your example, the `id` looks like it's going to be an `int`. Should I include "APP_ACCOUNT:" as a prefix to my `id`?

Also I'm seeing this:

[image]

Even though I did the clone successfully:

[image]

richo, 2025-02-17:
Hey,

Yeah sorry I wasn't sure if you just had two github accounts. If you are
seeing the APP_ACCOUNT: token then you should be sweet.

The error message you posted looks to me like you don't have the module
installed in the right place, but I think it's a different problem, could
you create a new issue to help keep them separate?

I am still frustrated that the hacs integration isn't sorted but I might
just yolo it today, it seems like it not being there is making more
problems than it solves.

thamindzzeye, 2025-02-18:
if I do the new pip command you mention i get this

`(env) ricky@Aragorn franklinwh-python % pip install -e . requirements.txt
Obtaining file:///Users/ricky/franklinwh-python
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Installing backend dependencies ... done
  Preparing editable metadata (pyproject.toml) ... done
ERROR: Could not find a version that satisfies the requirement requirements.txt (from versions: none)
HINT: You are attempting to install a package literally named "requirements.txt" (which cannot exist). Consider using the '-r' flag to install the packages listed in requirements.txt
ERROR: No matching distribution found for requirements.txt`

and i still get the same issue as @riddley   mentioned
`(env) ricky@Aragorn franklinwh-python % python3 bin/login.py r***.com test
Traceback (most recent call last):
  File "/Users/ricky/franklinwh-python/bin/login.py", line 7, in <module>
    from franklinwh import TokenFetcher
ModuleNotFoundError: No module named 'franklinwh'`

richo, 2025-02-18:
Oh sorry, you should't need to reference requirements.txt, it's just

    pip install -e .

That error is because python can't find my module, running that `pip` command basically installs a link to it inside the virtualenv.

thamindzzeye, 2025-02-19:
ok i have some progress. if i use a bad email account i get the 

`INFO:franklinwh.client:JSON payload: {'code': 1002, 'message': 'Account or password error!', 'result': None, 'success': False}`

BUT if i use my email and correct password i get this

`INFO:franklinwh.client:JSON payload: {'code': 1001, 'message': '请选择用户身份', 'result': None, 'success': False}`

that chinese message translates to 
"Please select a user role"

is there some user role i need to specify?

richo, 2025-02-19:
Thanks for getting back to me! That's super fun.

So uh, the mixed news is that I have no idea. This library was written by reverse engineering the behavior of the app, and I did not need to specify a role. I know you said your setup is pretty complicated, do you have more than one user configured in your account, or a reseller account or anything?

I can spend a little time trying to see what the app does in response to that error code and figure out what other context we need to provide. With that said, thanks so much for bearing with me, I think we can figure this out it just might take a bit of back and forth since I don't have a way to replicate this at my end.

david2069, 2025-02-25:
In the login request body there is userType tag with 0 must be normal user and likely 1 is an installler 

Example

--data-raw 'account=firstname.lastname %40email.com&lang=EN_US&optDevice=devicename&optSystemVersion=deviceOSversion&password=passwordhash&softwareVersion=APP1.10.0&type=1&userType=0' 

> Thanks for getting back to me! That's super fun.
>
> So uh, the mixed news is that I have no idea. This library was written by reverse engineering the behavior of the app, and I did not need to specify [...]

richo, 2025-02-28:
Ah sweet. Did you grab that with burp/httptoolkit or something similar? Or is there some easier way to poke at the app I'm not aware of?

david2069, 2025-03-01:
Yes, I have that and also have used  Proxyman on iOS and MacOS

They’re made it harder now with that latest apps. 😜

richo, 2025-03-01:
Hah, classic. I did the original research with httptoolkita and some static analysis of the android app.

I'll go add this parameter shortly and hopefully make this work properly.

david2069, 2025-03-06:
Hi Ricky

I enjoy your YouTube channel.

I saw you’re in Sydney for Everything Electric. I hope go Saturday.

Cheers,
David

> Hi Richo,
> 
> First of all, THANK YOU FOR THIS... I have a bit of a complex install, with SPAN and solar, 5 aPowers etc.. .
> 
> but yes I tried with the password in the secret yaml and also just directly in config.
> 
> since my password and id both contain letters and numbers should they too be in quotes in the config? like this
> 
> sensor:
> 
> * platform: franklin_wh
>   username: "[email@mail.com](mailto:email@mail.com)"
>   password: "somepassword123"
>   id: "12345A12W12345"
> 
> which items need quotes which dont? I'm pretty new to HA.
> 
> my SN for my aGate starts in 1013 ... I had the brain updated when I got my franklin + SPAN.
> 
> I'm a YouTuber (youtube.com/twobitdavinci) and a test bed for FranklinWH I have a lot of new products before they ship, happy to chat if you ever want to!

thamindzzeye, 2025-05-15:
that's awesome! look forward to meeting you again at the next show!

bdog4life, 2025-06-06:
Which config file am I supposed to edit

[image]

npdsomerhayes, 2025-06-06:
@bdog4life sorry I am travelling at the moment hopefully this will help

https://www.home-assistant.io/docs/configuration/

if you are still stuck I can try and give you more help tomorrow.

bdog4life, 2025-06-07:
Seems to have "worked" to get readings precisely once... not sure if there is something that needs to be configured for refreshing values.

npdsomerhayes, 2025-06-07:
No it should update every 30 seconds, I assume you have restarted HA?
Are there any errors in the logs?   Settings - System - Logs