richo/homeassistant-franklinwh #27: Feature Request: Smart Circuit Load Sensors

torrancew, 2025-03-07:
Hi @richo! You seem to have a good bit of experience with the Franklin APIs -- I was wondering if you knew how feasible it would be to report the Smart Circuit loads? I have a smart circuit that is otherwise *really* obnoxious to monitor (a bunch of mini-split-style heat pumps), and I was planning to install a CT clamp on the circuit but it seems like the Franklin stack has access to that data. I'm happy to run with it, build up a PR, etc -- but wanted to see if you had any thoughts/notes/pointers before I started down the tedius path of reversing HTTPS APIs

As always, thanks for building this! My install is only a few days old, but I get a ton of value out of your integration (and I don't know if I would have even gone with the Franklin in the first place if this plugin didn't exist!)

richo, 2025-03-07:
hey! I'm actually not positive that information is available to the aGate, I think it just aggregates everything into the "home load"  bucket? A good rule of them is that if you can see what you want in the app, we can probably make it available to homeassistant.

torrancew, 2025-03-07:
So, it is there but it is NOT obvious. I probably wouldn't have found it if I hadn't noticed it when the installer did the walk through 

For me, it's under
`Settings > Smart Circuits`, then hidden behind the nondescript arrow for each circuit

[image]

[image]

richo, 2025-03-07:
Huh!! Ok cool, let me see if I can find it.

Do you have a sense of what you're wanting this to look like in HA? Another sensor alongside home load?

torrancew, 2025-03-07:
Yeah, just a sensor per circuit would be fine for me, I'm happy to do
whatever other logic I need (summing two circuits up for a logical load,
etc) out of band

torrancew, 2025-03-11:
That's great! Looking at the results in my instance and comparing them to the app, I'm probably most interested in:

`SW1ExpPower` and `SW2ExpPower`. I can manually sum those up with helpers, and generate time-based energy sensors with PowerCalc.

@richo I know you have a lot of changes in-flight for this add-in, but I'd be happy to send a PR for this (when and if you're ready to accept it!)

richo, 2025-03-11:
I am always happy to accept PRs! I definitely have a lot of things in flight right now, and it's awesome to see a tiny little community building around this.

I definitely want to try to converge a bunch of the dangling branches I have to avoid a ton of gnarly conflicts down the line, but if you want to take a crack at this that would be awesome. The only real change I have that would directly be in conflict is that I'm noodling over how to make the interactor for Franklin a singleton that all the sensors and switches can share. right now the retry and backoff logic is really gnarly and fragile, but I haven't had much of a chance to figure out how that interacts with hass just loading addons into the main process.

torrancew, 2025-03-14:
I've had this patch live in my instance for a few days, and it seems to work well, so I've cut a PR for it. Let me know if you'd like to see any changes!

richo, 2025-03-14:
Skimming it looks good, I'd like to test it on my instance real quick but I think it looks good.

torrancew, 2025-03-14:
If you use V2L, I'd love to hear if it's working correctly!

torrancew, 2025-03-18:
I guess "implements" isn't one of the words GitHub auto-closes issues based on. I'm going to close this request -- I consider it delivered as of #28 getting merged.

closed 2025-03-18 by torrancew