richo/homeassistant-franklinwh #29: Not getting data from Franklin WH’s servers

ryanlim, 2025-05-06:
Today I started having some issues where Home Assistant was not getting fresh data from FranklinWH’s servers. Is this something isolated to just me or are others also seeing it?

This is the output from from Home Assistant’s logs.

```
home-assistant  | Traceback (most recent call last):                                                                                                        
home-assistant  |   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 960, in async_update_ha_state                                       
home-assistant  |     await self.async_device_update()                                                                                                      
home-assistant  |   File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1320, in async_device_update                                        
home-assistant  |     await hass.async_add_executor_job(self.update)                                                                                        
home-assistant  |   File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run                                                          
home-assistant  |     result = self.fn(*self.args, **self.kwargs)                                                                                           
home-assistant  |   File "/config/custom_components/franklin_wh/sensor.py", line 112, in update                                                             
home-assistant  |     stats = self._cache.fetch()                                                                                                           
home-assistant  |   File "/config/custom_components/franklin_wh/sensor.py", line 91, in fetch                                                               
home-assistant  |     self._fetch()                                                                                                                         
home-assistant  |     ~~~~~~~~~~~^^
home-assistant  |   File "/config/custom_components/franklin_wh/sensor.py", line 84, in _fetch
home-assistant  |     self.data = self.update_func()
home-assistant  |                 ~~~~~~~~~~~~~~~~^^
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 309, in get_stats
home-assistant  |     data = self._status()
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 264, in _status
home-assistant  |     data = self._mqtt_send(payload)['result']['dataArea']
home-assistant  |            ~~~~~~~~~~~~~~~^^^^^^^^^
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 357, in _mqtt_send
home-assistant  |     res = self._post(url, payload)
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 192, in _post
home-assistant  |     return retry(__post, lambda j: j['code'] != 401, self.refresh_token)
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 172, in retry
home-assistant  |     res = func()
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/franklinwh/client.py", line 190, in __post
home-assistant  |     res = requests.post(url, headers={ "loginToken": self.token, "Content-Type": "application/json" }, data=payload).json()
home-assistant  |           ~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/requests/api.py", line 115, in post                                 
home-assistant  |     return request("post", url, data=data, json=json, **kwargs)                    
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/requests/api.py", line 59, in request
home-assistant  |     return session.request(method=method, url=url, **kwargs) 
home-assistant  |            ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ 
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/requests/sessions.py", line 589, in request
home-assistant  |     resp = self.send(prep, **send_kwargs)
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/requests/sessions.py", line 703, in send
home-assistant  |     r = adapter.send(request, **kwargs)
home-assistant  |   File "/usr/local/lib/python3.13/site-packages/requests/adapters.py", line 700, in send
home-assistant  |     raise ConnectionError(e, request=request)
home-assistant  | requests.exceptions.ConnectionError: HTTPSConnectionPool(host='energy.franklinwh.com', port=443): Max retries exceeded with url: /hes-gate
way/terminal/sendMqtt (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x7a3ec966f750>: Failed to establish a new connection: [Errno -3] Try again'))
```

richo, 2025-05-06:
Yeah, so their API is really flaky and normally a handful of retries is enough but sometimes it isn't.

I have a [branch](https://github.com/richo/homeassistant-franklinwh/tree/caching-client) which I'm testing at my house which instead of letting homeassistant poll for data just tries once a minute and silently discards failure, which seems to be more reliable. In the really short term, if you restart home assistant it should clear (unless something else is wrong).

I expect to merge that branch sometimes this week or next.

richo, 2025-05-06:
Sure thing, let me know if it does. It's pretty reliable at home but it breaks just often enough that I wanted to have a more permanent solution. I just need to make the smart switch state use the same code and then if it's been stable at my house I'll merge it to main.

Thankyou for you the report!

ryanlim, 2025-05-06:
It seems at this point, I am no longer getting any data from Franklin’s servers :(

richo, 2025-05-06:
Fascinating. Did you change branches? Did you restart your home assistant?
Is it the same error?

ryanlim, 2025-05-06:
Yeah. I changed the branch and restarted.

Finally after 20 minutes, I got some data coming in, but I’m not sure how reliable this is.

I wonder if Franklin changed their API :/

richo, 2025-05-06:
It should take about a minute to prime but not 20. It's working ok here but
let me have a look.

ryanlim, 2025-05-06:
Just to give you an update. It seems after that 20 minutes or so, data finally started coming through consistently enough. I don't know if it was something FranklinWH was doing on their backend that was having issues.