richo/homeassistant-franklinwh #30: Load Switches

V1ru8, 2025-05-23:
I just tried the new load switches and some things seem off.

- Switch 1 & 2 always show almost the identical load. Switch 2 is always +2 watts compared to switch 1. This seems off, as I have two very different loads connected. (Wall Charger & HVAC)
- I saw the two switches are hard coded. Why is there no switch 3 load? (I tried adding one manually, but seems to not read any value)

richo, 2025-05-23:
Interesting, let me have a look and see what I see. Just to check, do you have your switches bonded? In our install switch 1+2 are combined.

V1ru8, 2025-05-23:
Ah yes, that is probably the case. As I have 2x50A load. And if I remember correct that's only possible with combining two switches. That would explain why there is no switch 3?
Also in the FWH app, each switch does show the right load.

richo, 2025-05-23:
The switch 3 thing is interesting and I actually don't have an answer, I didn't implement that feature, although I vaguely remember something about it. I will try to dig in and see if there's a better answer. I lowkey expect the answer is that switch 1 and 2 will show the same thing if they're bonded, and that switch 3 just needs something weird to work. If the app is able to show the same data then so is this module.

V1ru8, 2025-05-23:
What you're saying makes sense. Let me know if there is anything I can do to help figuring this out.

david2069, 2025-05-23:
The US version has two configuration options :smart-circuits-module-datasheetPDF Document · 1.5 MBOption A has x2 120V  (max 50 A) and x1 240Vc(80 A)Option B merges circuits 1 & 2  x2 240V 50 A & 80 A respectively The Australian/NZ model aGate is hardware version 1.2 and we only get option B - as we are 230/240V but at 65 amps  in totalsmart-circuits-module-datasheet-ausPDF Document · 221 KBSent from my iPhone

V1ru8, 2025-05-27:
I was looking into this a bit more and when I looked at the raw response from the API and there really is no switch 3, but the vlaues I found for `CarSW` matched exactly what I saw in the App for my Switch 3. 
```
{'CarSWCurr': 21,
 'CarSWExpEnergy': 71053,
 'CarSWImpEnergy': 0,
 'CarSWPower': 85,
…
```
So it seems that V2L and Switch 3 are overloaded, as I guess one can only use one or the other at the same time?
Either way, the the V2L sensors seem to have all I needed.

closed 2025-05-27 by V1ru8