richo/homeassistant-franklinwh #31: Not working with two batteries on one account

bayan9, 2025-05-26:
The configuration doesn't work with two separate batteries on one login

richo, 2025-05-26:
Two batteries? Or two aGates?

Two batteries should work, that's how mine is setup. With two aGates I *believe* it only look at the serial you included.

Can you share a little more about what's happening/if there are any errors?

bayan9, 2025-05-26:
> @ github. com>

I have two aGates. It's only loading one set of entities still, I t doesn't seem to have support for creating two sets of entities each tied to one serial number:

bayan9, 2025-05-27:
> https://github.com/richo/homeassistant-franklinwh/issues/31#issuecomment-2911131011
> )
>
>
> Ok, I got some answers from the developers but it's now pretty late here.
> I will try to get something out tomorrow.

Thank you

bayan9, 2025-06-30:
> On Mon, May 26, 2025 at 10:19 PM, richö butts < notifications@ github. com

Hi Richo, do you have any update?

srcLurker, 2025-07-02:
Thanks so much for creating this!

I also have multiple gateways (and multiple batteries on each gateway) and have created a PR based on the "multiple_gateways" branch. Please take a look when you have time and let me know if the PR looks OK or if there are changes desired.

The PR is: https://github.com/richo/homeassistant-franklinwh/pull/34

srcLurker, 2025-07-08:
PR#34 has been merged into main. This issue (in theory :) is resolved.

richo, 2025-07-08:
There were some bugs in that PR, I am working on them now.

richo, 2025-07-08:
Ok, what's on main is now working locally for me, and I believe you shoudl just need to set the `prefix` key in your config for each one.

rmazil, 2025-09-06:
Richo, 

I have 1 aGate and 2 batteries but Home Assistant is only showing 18kwh capacity.  This is the only configuration I added to yaml file. 
What am I missing to have two batteries show up?
sensor: 
  platform: franklin_wh
  username: "XXXXX"
  password: !secret franklinwh_password
  id: "1XXXXX"

rmazil, 2025-09-06:
> Ok I just cut `0.4.0` (And then immediately `0.4.1`, since I did a goof).
> 
> This should now work. You'll need to create a separate stanza for each of your installs, and set the `prefix` config key with a friendly name for each one.

Richo,

I have 1 aGate and 2 batteries but Home Assistant is only showing 18kwh capacity. This is the only configuration I added to yaml file.
What am I missing to have two batteries show up?
sensor:
platform: franklin_wh
username: "XXXXX"
password: !secret franklinwh_password
id: "1XXXXX"