richo/homeassistant-franklinwh #35: Two batteries are not showing up

rmazil, 2025-09-06:
How did you guys configure the yaml file to show two batteries?  For some reason Home Assistant is only showing 18kwh for my battery even though I have two apower2s charge to 100% so it should be showing almost 3kwh.  I added this to my yaml. I only added the serial number for the agate. 
sensor: 
  platform: franklin_wh
  username: "XXXX"
  password: !secret franklinwh_password
  id: "XXX"

rmazil, 2025-09-07:
> > How did you guys configure the yaml file to show two batteries?  For some reason Home Assistant is only showing 18kwh for my battery even though I have two apower2s charge to 100% so it should be showing almost 3kwh.  I added this to my yaml. I only added the serial number for the agate.
> > sensor:
> > platform: franklin_wh
> > username: "XXXX"
> > password: !secret franklinwh_password
> > id: "XXX"
> 
> As per the list of supported sensor entities - the only total provided is for the aGate battery state of charge (SOC) - which is a representation of all connected aPower batteries in as percentage.
> 
> There is no sensor for total capacity or available capacity in kWh - per aGate and connected batteries - if that what you mean.
> 
> There even no real-time running total of these battery kWh metrics even in FranklinWH. That as the historical metrics where you can find what are looking for.

I understand that.  I have one aGate and two aPower 2 connected to it.  My FranklinWH app on Android shows 30kWh capacity.  In home assistant with this plug-in, I see FranklinWH Battery Charge 18kWh (should be 30kWh, right?) and State of Charge shows 100%.

david2069, 2025-09-09:
As mentioned / this sensor Battery Charged is not the battery capacity - it is record accumulation of the daily battery kilowatts charged measured in kilowatt hours (kWh) stored per day. The Battery Discharged is the inverse.
 
It battery charged kWh can be different from the FHP app if your HA integration was down or lost connectivity for corresponding day. The HA integration  does not re/sync missing data." - it maybe missing.

If you aGate did not charge batteries (it was in Standby or Emergency Backup mode) for the whole day this figure will be zero.

 [image]

[image]